NMGRL/pychron #499: spectrum log scale

wcmcintosh, 2015-01-07:
When I plot an age spectrum, with K/Ca set to log, the scale does not appear to be a correct log scale
e.g. run 63392-01

jirhiker, 2015-01-07:
@wcmcintosh I believe this has been a long standing issue that is not unique to K/Ca but is related to the range.  If the range is less than X (maybe 1 or 2 log units) the graphical package doesn't render the ticks correctly. Also negative values maybe an issue. 

Is it an issue with the y axis ticks or actually not laying out the data points correctly.

jirhiker, 2015-01-08:
Neither of the above comments were applicable to the current issue. 
Spectrum._add_plot simply had the y scale hardcoded as "linear"

wcmcintosh, 2015-01-08:
I think that the data is being plotted on a log scale correctly, but it is a little hard to tell because only one tick mark is being shown. Manual tic interval selection is not appropriate for log scale. Auto selection should show a minimum of two tic marks.

jirhiker, 2015-01-09:
Need to make grid lines match ticks when log scale

closed 2015-08-05 by jirhiker