NMGRL/pychron #514: recall

wcmcintosh, 2015-01-23:
When I try to access recent Jan data from the current pycharm python version,
i get the following errors.The sample and analyses panes remain unpopulated.  Also, the date criteria is checked for 1/1/00

```
Exception occurred in traits notification handler for object: <pychron.processing.tasks.recall.recall_task.RecallTask object at 0x11bfd24d0>, trait: selected_projects, old value: [<pychron.envisage.browser.record_views.ProjectRecordView object at 0x11d3f47d0>], new value: [<pychron.envisage.browser.record_views.ProjectRecordView object at 0x11d3f4770>]
Traceback (most recent call last):
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/traits/trait_notifiers.py", line 340, in __call__
    self.handler( *args )
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_mixin.py", line 522, in _selected_projects_changed
    self._load_associated_labnumbers(names)
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_mixin.py", line 282, in _load_associated_labnumbers
    sams.extend(self._retrieve_recent_labnumbers(name))
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_mixin.py", line 303, in _retrieve_recent_labnumbers
    sams = self._retrieve_labnumbers()
  File "/Users/mcintosh/Programming/git/pychron/pychron/core/codetools/inspection.py", line 43, in dec
    return func(*args, **kw)
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_mixin.py", line 373, in _retrieve_labnumbers
    sams = progress_loader(ls, func)
  File "/Users/mcintosh/Programming/git/pychron/pychron/core/progress.py", line 83, in progress_loader
    return list(gen(progress))
  File "/Users/mcintosh/Programming/git/pychron/pychron/core/progress.py", line 74, in gen
    r = func(x, None, 0, 0)
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_mixin.py", line 371, in func
    return LabnumberRecordView(li)
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/record_views.py", line 26, in __init__
    self._create(dbrecord)
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/record_views.py", line 99, in _create
    level = pos.level
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/orm/attributes.py", line 316, in __get__
    return self.impl.get(instance_state(instance), dict_)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/orm/attributes.py", line 613, in get
    value = self.callable_(state, passive)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/orm/strategies.py", line 524, in _load_for_state
    return self._emit_lazyload(session, state, ident_key, passive)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/orm/strategies.py", line 563, in _emit_lazyload
    return loading.load_on_ident(q, ident_key)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/orm/loading.py", line 226, in load_on_ident
    return q.one()
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2310, in one
    ret = list(self)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2353, in __iter__
    return self._execute_and_instances(context)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2368, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 662, in execute
    params)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 761, in _execute_clauseelement
    compiled_sql, distilled_params
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 874, in _execute_context
    context)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1024, in _handle_dbapi_exception
    exc_info
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/util/compat.py", line 196, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 867, in _execute_context
    context)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/engine/default.py", line 324, in do_execute
    cursor.execute(statement, parameters)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/pymysql/cursors.py", line 117, in execute
    self.errorhandler(self, exc, value)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/pymysql/connections.py", line 189, in defaulterrorhandler
    raise errorclass, errorvalue
InternalError: (InternalError) (1054, u"Unknown column 'irrad_LevelTable.note' in 'field list'") 'SELECT `irrad_LevelTable`.id AS `irrad_LevelTable_id`, `irrad_LevelTable`.name AS `irrad_LevelTable_name`, `irrad_LevelTable`.z AS `irrad_LevelTable_z`, `irrad_LevelTable`.note AS `irrad_LevelTable_note`, `irrad_LevelTable`.holder_id AS `irrad_LevelTable_holder_id`, `irrad_LevelTable`.irradiation_id AS `irrad_LevelTable_irradiation_id`, `irrad_LevelTable`.production_id AS `irrad_LevelTable_production_id`, `irrad_LevelTable`.create_date AS `irrad_LevelTable_create_date`, `irrad_LevelTable`.last_modified AS `irrad_LevelTable_last_modified` \nFROM `irrad_LevelTable` \nWHERE `irrad_LevelTable`.id = %s' (246L,)
traits                                  : 2015-01-23 09:51:27,839 ERROR     (MainThread) Exception occurred in traits notification handler for object: <pychron.processing.tasks.recall.recall_task.RecallTask object at 0x11bfd24d0>, trait: selected_projects, old value: [<pychron.envisage.browser.record_views.ProjectRecordView object at 0x11d3f47d0>], new value: [<pychron.envisage.browser.record_views.ProjectRecordView object at 0x11d3f4770>]
Traceback (most recent call last):
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/traits/trait_notifiers.py", line 340, in __call__
    self.handler( *args )
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_mixin.py", line 522, in _selected_projects_changed
    self._load_associated_labnumbers(names)
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_mixin.py", line 282, in _load_associated_labnumbers
[... 29 lines cut ...]
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2353, in __iter__
    return self._execute_and_instances(context)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2368, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 662, in execute
    params)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 761, in _execute_clauseelement
    compiled_sql, distilled_params
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 874, in _execute_context
    context)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1024, in _handle_dbapi_exception
    exc_info
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/util/compat.py", line 196, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb)
  File "/Users/mcintosh/Library/Enthought/Canopy_64bit/User/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 867, in _execute_context
    context)
[... cut ...] lTable`.z AS `irrad_LevelTable_z`, `irrad_LevelTable`.note AS `irrad_LevelTable_note`, `irrad_LevelTable`.holder_id AS `irrad_LevelTable_holder_id`, `irrad_LevelTable`.irradiation_id AS `irrad_LevelTable_irradiation_id`, `irrad_LevelTable`.production_id AS `irrad_LevelTable_production_id`, `irrad_LevelTable`.create_date AS `irrad_LevelTable_create_date`, `irrad_LevelTable`.last_modified AS `irrad_LevelTable_last_modified` \nFROM `irrad_LevelTable` \nWHERE `irrad_LevelTable`.id = %s' (246L,)

```

wcmcintosh, 2015-01-23:
I notice reference to fairly old irradiation levels in the final line of the errors

jirhiker, 2015-01-23:
InternalError: (InternalError) (1054, u"Unknown column 'irrad_LevelTable.note' in 'field list'")

The latest version of Pychron is out of sync with the database. Database needs to be updated.
Im surprised it didn't give a warning at startup saying database out of date

closed 2015-01-23 by jirhiker