NMGRL/pychron #530: Obama CDD switch off

wcmcintosh, 2015-03-10:
Obama is currently chronically suffering from CDD switchoff every few analyses
This may be a hardware problem, or a Qtegra problem, but the timing is very suspicious
Most times, switchoff occurs at the beginning of the peak center, after successfully collecting peak and baseline data.  
Given this timing, is there any way that this behavior could somehow be related to Pychron???
Could there be something about the mass move that is somehow ringing a large beam (N2? H2O?) onto the CDD?  I am at a loss on this one, but it seems like Qtegra should have no way of consistently shutting down the CDD between peak measurements and centering.

wcmcintosh, 2015-03-11:
In a situation like this, where valid data is collected but not peak center data, would it be possible to save the partial data set rather than losing it?

jirhiker, 2015-03-11:
As always the raw data is stored locally in a special data file. this data file is written too every iteration so it has a complete copy of the intensity vs time data. This has been a feature since the earliest days of pychron. 

This file could be used to import the data post crash. Mass spec does this automatically (or at least asks user it they want to fix/import crashed run). Ive just never fully implemented importing crashed analyses.

Its theoretically possible to save the intensity data to the db prior to starting a peak center but this method is not as flexible as I would want. I could also add a pyscript command to explicitly trigger a "save intensity data to db" event, but this would require modification to the existing pyscripts.

jirhiker, 2015-03-11:
added a 300 second timeout to peak center. possible fix leaving instrument static and losing data.

jirhiker, 2015-03-16:
wcmcintosh 

after changing the baseline position from 33.2 to 39.5, the Obama CDD problem did not appear for the rest of the analyses in the experiment. After the last quit, before the baseline change, Matt scanned the instrument and found only tiny beams from masses 15 to 50, so ringing a large beam onto the CDD doesn't seem likely. Also, the last CDD quit resulted in an unchecked CDD box, but not a yellow CDD disabled button. Most other quits disabled and unchecked the CDD.

Not sure what's going on here. I'd like to discuss current large mass move procedures.

closed 2015-08-05 by jirhiker