NMGRL/pychron #576: ideogram - manual grouping

wcmcintosh, 2015-07-30:
some options to manually modify the run list for the pipeline would be desirable

e.g. In some cases it would be helpful to manually group and/or modify the order of analyses in the sample list, as well as to be able to tag and eliminate specific analyses

jirhiker, 2015-07-31:
Need to elaborate.

* Which run list are you referring to, Browser Analyses table or the Pipeline Analyses table. 
* reordering analyses is mostly pointless though it's possible in the Pipeline Analyses table.
* Analyses may be deleted from either table. 
* Tagging can be down from the Pipeline Analyses table. It will never be done from the Browser View.

wcmcintosh, 2015-07-31:
Reordering and manual group markers are special purpose tasks, for example comparing a and b steps on an ideograms. Not completely pointless.  Where would this occur? Perhaps in its own special node?  The manual arrangement node, added to the pipeline only when wanted?  Is that I'm the true pipeline spirit?

William C McIntosh

jirhiker, 2015-07-31:
add a group by step option to GroupingNode

wcmcintosh, 2015-07-31:
Jake:

	This would be great.  I often use this approach to examine data, although it doesn’t generally end up in publications.  Both Pychron and Mass Spec lack a slick automated way to sort the data in this fashion.

	In addition, how about a manual sorting/grouping option as well for special situations?  Not unlike the review option to manually reject data points.

Bikll

William McIntosh
Geochronologist/Volcanologist
NM Bureau of Geology
Associate Professor
Earth and Environmental Science Dept
NM Tech
801 Leroy Place
Socorro, NM 87801
575-835-5271
mcintosh@nmt.edu

jirhiker, 2015-07-31:
analyses can be reordered by dragging.
group a selected set of analyses by right clicking.

wcmcintosh, 2015-07-31:
Jake:

	Beautiful.  I’ll try it out

Bill

William McIntosh
Geochronologist/Volcanologist
NM Bureau of Geology
Associate Professor
Earth and Environmental Science Dept
NM Tech
801 Leroy Place
Socorro, NM 87801
575-835-5271
mcintosh@nmt.edu

closed 2015-08-05 by jirhiker