NMGRL/pychron #595: blank accounting DVS

wcmcintosh, 2015-08-13:
for putting the data into DVS format, how about going back x blanks and forward y blanks, as an alternative to basing the choice on time?

jirhiker, 2015-08-13:
Ive imported irradiations 258-269. I will continue with the current time search for the remainder of the irradiations. 

your suggested approach is good in theory but I think "want"lessly complicates the import process.

closed 2015-08-13 by jirhiker

wcmcintosh, 2015-08-13:
Hi Jake:

	I thought that it might simplify things.

Bill

William McIntosh
Geochronologist/Volcanologist
NM Bureau of Geology
Associate Professor
Earth and Environmental Science Dept
NM Tech
801 Leroy Place
Socorro, NM 87801
575-835-5271
mcintosh@nmt.edu