NMGRL/pychron #604: Inconsistent results using Touptek Camera

lefsky, 2015-09-05:
I have been obtaining inconsistent but mostly negative results using pychron to interface to a touptek camera. For instance, I can retrieve some basic information (i.e. serial number) but not others (e.g. size of image) and I can't get it to snap a still.  Is this part of the software still in a development phase?

jirhiker, 2015-09-05:
I haven't worked on ToupCamCamera class for a few months but when I last used it most features were working. 

If I remember correctly, tho, ToupCamCamera.save was not working. In order to save a still use ToupCamCamera.get_pil_data()
e.g
```
cam = ToupCamCamera()
cam.open()
im = cam.get_pil_data()
with open('snapshot.jpg', 'w') as wfile:
    im.save(wfile, 'JPEG')
```

lefsky, 2015-09-05:
Thanks for the reply. Can you tell me what hardware you were running this code on? I have had terrible luck with the Toupcam I have depending on which USB hardware I was using. Looks like it was a mac based on the use of dylib.

jirhiker, 2015-09-06:
Im using MacBook Pro (mid 2014) OSX 10.10.4 with an OMAX A35100U3 10MP USB 3.0 camera

lefsky, 2015-09-06:
And that's a touptek camera under another brand name?

jirhiker, 2015-09-07:
Yea. Its a Touptek U3CMOS series.

lefsky, 2015-09-07:
Jake,

Thanks again.

I am going to be looking for a programmer that has experience in python to
work on the software needed collect and process images from an aerial film
scanner of my own design. Briefly, the scanner consists of an array of
cameras that are used to scan 9 x 9 inch aerial photos at a resolution of
6.5 um. The camera array needs to be calibrated for focus, radiometry and
geometric position and the software needs to integrate the motion of the
camera array in 2 dimensions as well as controlling film advance and
holding mechanisms. It then needs to position and mosaic the individual
images to create a single image.

I have no idea if you pursue programming outside of your Ar-Ar
geochronology research, but you seem to be perfect for this type of job.

M

lefsky, 2015-09-10:
Jake,

I'd rather be talking high level design of the application I envisage, but
right now I could use some practical advice.

I have been an awful time getting the toupcam to work on any platform other
than windows (and even then it is inconsistent). I am trying it on a
macbook with osx 10.9.5 as this is the closest I have to your hardware.
When I run ToupLite, it starts and allows me to select the camera but the
stalls and gives me an error window "Cannot preview, please re-plug the
device and try again". Needless to say, re-plugging it doesn't work.

Have you had similar problems? I've contacted Touptek, but they have
nothing to suggest.

M

jirhiker, 2015-09-11:
I've used ToupViewX and ToupLite and haven't seen the error you mentioned. Sounds a bit more like a camera hardware problem than software if you're having trouble on various platforms.  Are you going directly from the computer to the camera or is there a usb hub in between?

lefsky, 2015-09-11:
Directly- this is a new camera model so the hardware problem is possible.

closed 2016-04-17 by jirhiker

jirhiker, 2016-06-20:
A separate toupcam project has been created at github.com/NMGRL/toupcam. 

This project should be used instead of pychron if you want to use ToupTek cameras with python