NMGRL/pychron #605: IC factor plot

wcmcintosh, 2015-09-09:
IC factor plot has order-of-magnitude too high errors on H1/CDD factor relative to mass spec

compare A-01-J-2519 (same for all airs examined)
pychron ±2.39%
mass spec ±0.35%

attempting to right-click recall from screen gave no recall and following error

```
Exception occurred in traits notification handler for object: <pychron.pipeline.engine.PipelineEngine object at 0x11a5e97d0>, trait: recall_event, old value: <undefined>, new value: [<pychron.dvc.dvc_analysis.DVCAnalysis object at 0x121339170>]
Traceback (most recent call last):
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/traits/trait_notifiers.py", line 520, in _dispatch_change_event
    self.dispatch( handler, *args )
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/traits/trait_notifiers.py", line 483, in dispatch
    handler( *args )
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 279, in _handle_analysis_tagging
    self.recall(new)
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_task.py", line 181, in recall
    self._open_recall_editors(ans)
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_task.py", line 290, in _open_recall_editors
    av = rec.analysis_view
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis.py", line 329, in analysis_view
    mod = __import__(mod, fromlist=[klass, ])
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/view/analysis_view.py", line 25, in <module>
    from pychron.processing.analyses.view.error_components_view import ErrorComponentsView
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/view/error_components_view.py", line 26, in <module>
    from pychron.canvas.canvas2D.piechart_canvas import PieChartCanvas
  File "/Users/mcintosh/Programming/git/pychron/pychron/canvas/canvas2D/piechart_canvas.py", line 21, in <module>
    from pychron.canvas.canvas2D.scene.pie_chart_scene import PieChartScene
  File "/Users/mcintosh/Programming/git/pychron/pychron/canvas/canvas2D/scene/pie_chart_scene.py", line 24, in <module>
    from pychron.canvas.canvas2D.scene.primitives.primitives import QPrimitive
  File "/Users/mcintosh/Programming/git/pychron/pychron/canvas/canvas2D/scene/primitives/primitives.py", line 34, in <module>
    import Image as PImage
ImportError: No module named Image
traits                                  : 2015-09-09 17:05:50,275 ERROR     (MainThread) Exception occurred in traits notification handler for object: <pychron.pipeline.engine.PipelineEngine object at 0x11a5e97d0>, trait: recall_event, old value: <undefined>, new value: [<pychron.dvc.dvc_analysis.DVCAnalysis object at 0x121339170>]
Traceback (most recent call last):
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/traits/trait_notifiers.py", line 520, in _dispatch_change_event
    self.dispatch( handler, *args )
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/traits/trait_notifiers.py", line 483, in dispatch
    handler( *args )
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 279, in _handle_analysis_tagging
    self.recall(new)
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_task.py", line 181, in recall
    self._open_recall_editors(ans)
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_task.py", line 290, in _open_recall_editors
    av = rec.analysis_view
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis.py", line 329, in analysis_view
    mod = __import__(mod, fromlist=[klass, ])
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/view/analysis_view.py", line 25, in <module>
    from pychron.processing.analyses.view.error_components_view import ErrorComponentsView
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/view/error_components_view.py", line 26, in <module>
    from pychron.canvas.canvas2D.piechart_canvas import PieChartCanvas
  File "/Users/mcintosh/Programming/git/pychron/pychron/canvas/canvas2D/piechart_canvas.py", line 21, in <module>
    from pychron.canvas.canvas2D.scene.pie_chart_scene import PieChartScene
  File "/Users/mcintosh/Programming/git/pychron/pychron/canvas/canvas2D/scene/pie_chart_scene.py", line 24, in <module>
    from pychron.canvas.canvas2D.scene.primitives.primitives import QPrimitive
  File "/Users/mcintosh/Programming/git/pychron/pychron/canvas/canvas2D/scene/primitives/primitives.py", line 34, in <module>
    import Image as PImage
ImportError: No module named Image

```

jirhiker, 2015-09-10:
from the Terminal
```
conda install PIL
```

jirhiker, 2015-09-10:
Not sure where to go with the error mismatch, appears to be more of a philosophical issue than a programming bug in pychron.
```
Ar40 = 80.84+/-0.15
Ar36 = 0.27+/-0.006

Ar40/Ar36 = 299.41+/-6.68  or 2.23% 
```
Include/exclude blank and/or baseline errors has little effect. 

The ratio error is dominated by the error in 36 so maybe mass spec is not propagating this error i.e
```
 A40/A36 = (80.84+/-0.15) / 0.27
         = 299.41+/-0.55 (0.19%)
```

closed 2015-09-10 by jirhiker

wcmcintosh, 2015-09-10:
Hi Jake:

	Michael was looking for you about 1:45.
	
	Also, termination does not seem to be working.

	Truncation is working fine. So I’ll modify the conditional file accordingly.

Bill

William McIntosh
Geochronologist/Volcanologist
NM Bureau of Geology
Associate Professor
Earth and Environmental Science Dept
NM Tech
801 Leroy Place
Socorro, NM 87801
575-835-5271
mcintosh@nmt.edu