NMGRL/pychron #645: IC fit/general comment

wcmcintosh, 2015-09-23:
It seems that we use the convenient 2 sigma filtering of iso ev data quite regularly.  It might be appropriate to allow similar filtering of other forms of data, such as IC fit, in part to add consistency to treatment of the data, rather than starting with user-selected rejection of data points.  Perhaps this or at least some kind of filtering should be available for more forms of data, perhaps even age data on ideograms.  It is currently applied (in a form) to age spectra.  Nothing critical here, just thinking.

wcmcintosh, 2015-09-23:
Along the same lines:  First - thanks for adding all of the fit options.  The review node makes it very easy to play around with various fits.  I am wondering if there is a bug in the SEM fit.  SEM errors are very tiny, near zero.  For the data that I am fitting (NM-273), I have 11 air analysis data points.  Am I correct that SEM should roughly equal SD divided by sqrt 11?  It looks like SEM is much less than 1/3 of the SD value.

wcmcintosh, 2015-09-23:
Comment on Bracketing-Interpolate fit fo ICFactors.  If there are no airs prior to the first unknowns, the program apparently interpolates between zero and the first air.  It would probably be better to extrapolate backwards from the data, rather than use zero as an endpoint.

wcmcintosh, 2015-09-23:
For IC (and other) fits, could "Following" also be an option?  Not something I use much, but it is handy in cases of missing or deranged prior data points.

jirhiker, 2015-09-23:
When doing a least squares regression (fitting linear, parabolic etc) SD and SEM are not referring to the traditional "Standard Deviation" and "Standard Error of the Mean" definitions. We've discussed this numerous times

SEM = Standard_Fit_Error * Sqrt(Variance_Y)
where Standard_Fit_Error = Sqrt( Sum_of_residuals / (N-dof))

SD = quadratic sum of Standard_Fit_Error and SEM

In this case SEM is the standard error for a predicted mean value at Y at a given X, SD is the predicted error of an individual observation at X

jirhiker, 2015-09-23:
Added Filter outliers BlanksOptions. 

**NEED TO SET OPTIONS TO FACTORY DEFAULTS TO WORK CORRECTLY**
otherwise you will generate an exception

Use the  [image] button to set factory defaults

closed 2015-10-26 by jirhiker