NMGRL/pychron #648: pipeline ideogram-significant digits

wcmcintosh, 2015-09-23:
How are significant digits in the mean calculation set? In the calculations tab, could the user choose automatic or manual setting of the significant digits?  Al's automatic setting of significant digits in figures was nice up to a point, but there were many many circumstances where I has to modify them after the fact in illustrator.

jirhiker, 2015-09-23:
sig figs for mean are set in IdeogramOptions/Display. 

there are no plans to implement an automatic sig fig determination.  However if sig figs is set to 1 and the error is 0.01, 0.01 will be displayed instead of 0.0

wcmcintosh, 2015-09-23:
I must be missing something here, but I don't see IdeogramOptions/Display when I click the ideogram node.

wcmcintosh, 2015-09-23:
sorry, just found it

wcmcintosh, 2015-09-23:
works perfectly

closed 2015-09-23 by jirhiker