NMGRL/pychron #649: pipeline ideogram - tabs

wcmcintosh, 2015-09-23:
lower priority pipeline ideogram click-reduction enhancement request

would it be possible to right click to save rejected points as your choice of either omit or invalid tabs?

jirhiker, 2015-09-26:
this will likely not be implemented. Trigger a tagging event via right click does not fit well with the current model. 

Use the toolbar actions Tag or Set Invalid instead

closed 2015-09-26 by jirhiker

wcmcintosh, 2015-09-29:
Jake:

	I feel that I should push back a little on this one.  I think that the tagging and persistence  of tagging is a current weak area in pychron, from a user standpoint.

Bill

William McIntosh
Geochronologist/Volcanologist
NM Bureau of Geology
Associate Professor
Earth and Environmental Science Dept
NM Tech
801 Leroy Place
Socorro, NM 87801
575-835-5271
mcintosh@nmt.edu

jirhiker, 2015-09-29:
What is your suggestion? Setting invalid or any other tag is available by right-clicking on individual analyses. Right-clicking in the white space to set tags for all temporarily omitted analyses is not easy and not something I feel is a useful use of time. Therefore you can use the Toolbar actions Set Tag or Set Invalid to accomplish the same thing. 

I take great exception to blanket statements such as "I think that the tagging and persistence  of tagging is a current weak area in pychron, from a user standpoint," especially because you convulting multiple issues.

wcmcintosh, 2015-09-29:
Jake:

	I apologize if I have caused you to take exception.

	I’m just being honest about what I feel is a glitch in a generally smooth workflow.  It comes down to data filtering in the end.  Ideally, the choices to include or exclude data would become much less arbitrary, which I guess means more automatic and based on some stated criteria, along the lines of plateau criteria.  You might have insight on how to use an AI approach to weed out inaccurate data.  Removing points outside of 2 sigma is potentially a step in the right direction, but it is a pretty dull knife.

	Until we can more effectively make sensible and reproducible choices about which data belongs in the data set, and which should be omitted, what we have available is the user selected data set.  I feel like pychron makes it easy to graphically select data to omit from data sets, but makes it more difficult (i.e. more clicks, more windows) than necessary to “commit” these choices. Going to the toolbar is great for making special choices (e.g. omit from weighted mean, not from isochron), but for 98% of our data situations, a very simple way to visualize, tag, and commit invalid and omit data is a very good thing for the user.  

	My picture of how this would work is just to have 2 choices in the right click window - set selected analyses as invalid, and set selected analyses as omit.  Perhaps there are alternative approaches to right-clicking on white space.  Could it be available as an option for the entire selected data set even when clicking on one point?  

	Maybe I am missing something in the work flow that makes this easy, or maybe there is a better way to make and save choices.

Bill

	 

  
William McIntosh
Geochronologist/Volcanologist
NM Bureau of Geology
Associate Professor
Earth and Environmental Science Dept
NM Tech
801 Leroy Place
Socorro, NM 87801
575-835-5271
mcintosh@nmt.edu