NMGRL/pychron #651: pipeline ideogram question

wcmcintosh, 2015-09-23:
When one right-clicks and chooses to export a CSV file, what value is being saved in the Y0 column?

jirhiker, 2015-09-23:
the relative probability. Y0 is the current relative probability (i.e solid curve) and Y1 is the original relative probability (i.e dashed curve)

wcmcintosh, 2015-09-23:
That seems sensible, but for some reason there are integers or negative numbers among the YO values in some cases.

Here are values for 24162 and 24163, no excluded points
28.08338882	1
28.14396726	2
28.15518491	3
28.18078664	4
28.19774496	5
28.19969472	6
28.21283551	7
28.26345718	8
28.27427649	9
28.28732407	10
28.2926094	11
28.30420488	12

Here are values for the same data set with some excluded points

x0	y0
28.08338882	92.25022731
28.14396726	-62.1477962
28.15518491	47.90729974
28.18078664	31.03256248
28.19774496	-16.26101033
28.19969472	109.9610081
28.21283551	92.59058449
28.26345718	89.24041287
28.27427649	226.7671715
28.28732407	107.6759051
28.2926094	70.34757199
28.30420488	82.52643379

closed 2015-09-24 by jirhiker