NMGRL/pychron #655: Spectrum question and observations

wcmcintosh, 2015-09-24:
What is the status of the spectrum plotting in the pipeline?

Are tickets welcome on spectra, or is it not yet ready?  I get the impression that some of the options panes were adapted from ideogram routines, but are not yet fully implemented.  I'm happy to comment on current observations if that would be helpful or to wait until later if that is more appropriate.

Initial observations:
plotting of data, axes, integrated age, plateau age OK
auxiliary plots OK

spectrum options: 

main tab:
 -scale log/linear OK
 -graph height OK
 -label option OK
 -x error not relevant?
 -y errors shown regardless of option selected
 -y tick option OK
 -y label no effect

spectrum tab:
  -empty

appearance tab:
 - grid visibility checkboxes not present
 -other features OK
 -still hoping someday for option to turn off central line, or make it transparent

calculations tab:
 -all features OK

display tab:
 -labeling steps by power OK (not sure about step, no step data available)
 -significant figures and labeling options in plateau and integrated age OK
 -plateau arrowheads OK
 -legend OK

groups tab:
-all functions OK

wcmcintosh, 2015-09-24:
title option would be useful

wcmcintosh, 2015-09-24:
spectrum options:
 - all existing plot option font settings work fine
 - font/size option for tic labels needed

wcmcintosh, 2015-09-24:
spectrum options:
 - marker type and marker size not needed, holdover from ideogram

jirhiker, 2015-09-24:
Started updating spectrum options
1. removed x errors
2. center line style (including No Line) and center line width added to Groups

an explicit list of what is missing would be nice

closed 2016-01-17 by jirhiker