NMGRL/pychron #664: problem running obama/diode unknowns

nmgrluser, 2015-10-07:
airs running ok on diode, but not unknowns

pychron seems to want experiment identifier, but dialog to add same is greyed out
overcame problem by editing experiment file adding "test" in experiment identifier field

now getting following error when unknown begins to run

is do_extraction expecting multiple arguments, not just duration?

'''221-hole                                 -- invalid stage map file. /Users/obama/Pychron_dev/setupfiles/tray_maps/221-hole.txt. Problem with line 3: 3,119,219,103,111

AutomatedRun                             -- No value "sensitivity_multiplier" in metadata
obama_diode.py                           -- _execute: Traceback (most recent call last):
  File "/Users/obama/Programming/git/pychron/pychron/pyscripts/pyscript.py", line 389, in execute_snippet
    func(*argv)
  File "<string>", line 43, in main
  File "<string>", line 72, in do_extraction
  File "/Users/obama/Programming/git/pychron/pychron/pyscripts/pyscript.py", line 130, in decorator
    raise PyscriptError(fname, 'invalid arguments count for {}, args={} kwargs={}'.format(fname, args, kw))
PyscriptError: Pyscript error in begin_interval

invalid arguments count for begin_interval, args=(40.0,) kwargs={}

AutomatedRun                             -- Invalid script syntax for "obama_diode.py"
ExperimentExecutor                       -- extraction did not complete successfully
ExperimentExecutor                       -- automated runs did not complete successfully
ExperimentExecutor                       -- error: Extraction Failed
'''

jirhiker, 2015-10-08:
Version issues with the obama/diode version were discovered during work in Manitoba. Updating may fix the issue. 

I actually was going to advice not running obama until I returned.  sorry. 

those errors may be spurious. I will examine asap.

jirhiker, 2015-10-08:
begin_interval error was not spurious and has been fixed in current version. @wcmcintosh may update the obama version via PyCharm. 

One small note. Use three backticks to enclose error messages. 
the correct format is 
\`\`\`
error text
\`\`\`

here is a correctly formatted error message
```
AutomatedRun                             -- No value "sensitivity_multiplier" in metadata
obama_diode.py                           -- _execute: Traceback (most recent call last):
  File "/Users/obama/Programming/git/pychron/pychron/pyscripts/pyscript.py", line 389, in execute_snippet
    func(*argv)
  File "<string>", line 43, in main
  File "<string>", line 72, in do_extraction
  File "/Users/obama/Programming/git/pychron/pychron/pyscripts/pyscript.py", line 130, in decorator
```
not 
```AutomatedRun                             -- No value "sensitivity_multiplier" in metadata
obama_diode.py                           -- _execute: Traceback (most recent call last):
  File "/Users/obama/Programming/git/pychron/pychron/pyscripts/pyscript.py", line 389, in execute_snippet
    func(*argv)
  File "<string>", line 43, in main
  File "<string>", line 72, in do_extraction
  File "/Users/obama/Programming/git/pychron/pychron/pyscripts/pyscript.py", line 130, in decorator
```
or 
'''AutomatedRun                             -- No value "sensitivity_multiplier" in metadata
obama_diode.py                           -- _execute: Traceback (most recent call last):
  File "/Users/obama/Programming/git/pychron/pychron/pyscripts/pyscript.py", line 389, in execute_snippet
    func(*argv)
  File "<string>", line 43, in main
  File "<string>", line 72, in do_extraction
  File "/Users/obama/Programming/git/pychron/pychron/pyscripts/pyscript.py", line 130, in decorator
'''

If there is any question as to whether the formatting is correct remember to use the "Preview" tab. 
Correctly formatted error messages will be in a monospace font and have a light grey background.

jirhiker, 2015-10-08:
UMan changes where just merged with NMGRL so issues may exist if you update obama's version. 

Its worth updating obama and testing but don't be surprised if Errors/Exceptions are prevalent. My main concern is saving to Mass Spec database but issues could appear at any point in the process

closed 2015-10-10 by jirhiker