NMGRL/pychron #666: diode error on startup

wcmcintosh, 2015-10-08:
```
/Users/obama/anaconda/bin/python /Users/obama/Programming/git/pychron/launchers/pyexperiment_debug.py
Archiver                                : 2015-10-08 08:28:11,472 INFO      (MainThread) Files older than 2015-09-24 08:28:11.472010 will be archived
Archiver                                : 2015-10-08 08:28:11,473 INFO      (MainThread) Archives older than 1 months will be deleted
Archiver                                : 2015-10-08 08:28:11,473 INFO      (MainThread) Archive cleaning complete
Traceback (most recent call last):
  File "/Users/obama/Programming/git/pychron/launchers/pyexperiment_debug.py", line 23, in <module>
    entry_point('pyexperiment', 'PyExperiment', '_dev', debug=True)
  File "/Users/obama/Programming/git/pychron/launchers/helpers.py", line 825, in entry_point
    user = initialize_version(modname, debug)
  File "/Users/obama/Programming/git/pychron/launchers/helpers.py", line 997, in initialize_version
    from pychron.core.helpers.exception_helper import set_exception_handler, report_issues
  File "/Users/obama/Programming/git/pychron/pychron/core/helpers/exception_helper.py", line 24, in <module>
    import keyring as keyring
ImportError: No module named keyring

Process finished with exit code 139

```

wcmcintosh, 2015-10-08:
missing keyring module

wcmcintosh, 2015-10-08:
found and installed keyring using terminal

jirhiker, 2015-10-08:
Excellent. another method is to use pycharm.

Did you use pip or conda from the terminal? just curious

Install packages via PyCharm

1. goto pycharm.
2. navigate to the file pychron/core/helpers/exception_helper.py

3. near the top of the file you should see
```
# ============= standard library imports ========================
import keyring as keyring
```
place the cursor on the line
```
import keyring as keyring
```

Hit Cmd+1
select "Install package keyring"

wcmcintosh, 2015-10-08:
Jake:

From the terminal, I used massspec:~ obama$ easy_install keyring

I tried your pycharm approach just now, but am not getting a response from cmd+1

We are still having the same problem with the Diode script, specifically Do_extraction
same error as before

Also (I know that this shouldn't be in this ticket), we are unable to access or ping the server at 129.138.12.131

Lisa has a connection open on her computer that works, but we can't establish any new connections

The remote desktop window suggests that the server now at 192.168.1.1 , but it is neither pingable or controllable at this address either.  

A possibly related symptom is that our local VPN is not working.  Ideas?

wcmcintosh, 2015-10-08:
Here is the error, is do extraction looking (or getting??) multiple arguments for Do_extraction?:

```
221-hole                                 -- invalid stage map file. /Users/obama/Pychron_dev/setupfiles/tray_maps/221-hole.txt. Problem with line 3: 3,119,219,103,111

AutomatedRun                             -- No value "sensitivity_multiplier" in metadata
ExperimentExecutor                       -- extraction did not complete successfully
ExperimentExecutor                       -- automated runs did not complete successfully
ExperimentExecutor                       -- error: User Canceled
AutomatedRun                             -- No value "sensitivity_multiplier" in metadata
obama_diode.py                           -- _execute: Traceback (most recent call last):
  File "/Users/obama/Programming/git/pychron/pychron/pyscripts/pyscript.py", line 389, in execute_snippet
    func(*argv)
  File "<string>", line 43, in main
  File "<string>", line 72, in do_extraction
  File "/Users/obama/Programming/git/pychron/pychron/pyscripts/pyscript.py", line 84, in decorator
    args, kw))
PyscriptError: Pyscript error in obama_diode.py

invalid arguments count for begin_interval, args=(40.0,) kwargs={}

AutomatedRun                             -- Invalid script syntax for "obama_diode.py"
ExperimentExecutor                       -- extraction did not complete successfully
ExperimentExecutor                       -- automated runs did not complete successfully
ExperimentExecutor                       -- error: Extraction Failed
```

jirhiker, 2015-10-08:
OK next time use pip instead of easy_install. (EasyInstall is just an outdated package manager thats been replaced)

The cause for the error is esoteric and has more to due with error checking and script duration calculation. update pychron and see if that fixes it.

wcmcintosh, 2015-10-08:
And how about the other errors?  What can I do to address them, just for my own mental health?:

```
221-hole                                 -- invalid stage map file. /Users/obama/Pychron_dev/setupfiles/tray_maps/221-hole.txt. Problem with line 3: 3,119,219,103,111

AutomatedRun                             -- No value "sensitivity_multiplier" in metadata
```

wcmcintosh, 2015-10-08:
Updated and ran. Same esoteric error.

wcmcintosh, 2015-10-08:
Just noticed error displayed in pychron last round

```
root                                    : 2015-10-08 10:16:16,423 CRITICAL  (MainThread) File "/Users/obama/anaconda/lib/python2.7/site-packages/pyface/ui/qt4/timer/do_later.py", line 52, in Notify
    def Notify(self):

```

wcmcintosh, 2015-10-08:
Updated.  Seems to be working.  Laser firing.  We will keep you posted.

jirhiker, 2015-10-08:
sorry previous update was a mistype. try now. 

invalid stage map file is not a critical error. if you want to fix it compare the 211-hole.txt file to other tray maps. the correct format is
```
circle, 1.25
1,2,3
1,2,3
x,y
x,y
x,y
```

lines 2 and 3 are not relevant for the current version
line 2= holes to draw
line 3= auto calibration holes

also make sure that the file has Unix (LF) line endings not Mac (CR). To check open in TextWrangler. There is a dropdown at the bottom of the window for setting line endings

No value "sensitivity_multiplier" is a warning not an error

wcmcintosh, 2015-10-08:
Diode seemingly working now.  Will update again when this run is done.

Fixed the 221 file.

Does the sensitivity warning mean something?  I am uncertain how Pychron handles sensitivity.

wcmcintosh, 2015-10-08:
How is Canada treating you?

wcmcintosh, 2015-10-08:
I must have caught your last update before you sent the message.  All files up to date.  Diode running OK I think.  The mass spec database seems to be updating OK.

One other oddball symptom on Obama yesterday.  Centering view had many centering runs drawn over each other,  Seems OK today.

jirhiker, 2015-10-08:
The sensitivity warning means that a "sensitivity_multiplier" is not specified in the extraction script. 

The idea is that there is a nominal sensitivity that gets multiplied by the sensitivity_multiplier to determine the actual sensitivity for a given extraction line configuration. e.g the diode might have a sensitivity_multiplier of 0.25 and the co2 a sensitivity of 0.5, essentially indicated the diode extraction configuration has 2x the volume compared to the co2. 

This was your concept!

However pychron doesn't really do anything with sensitivity at this point so significant work is required. This is why when you plot Mol of 39 in an ideogram you are actually plotting fA.

wcmcintosh, 2015-10-08:
Thanks Jake.  Should we put in sensitivity_multiplier = 1.0 placeholders in the extraction scripts for good mental health?

We restarted 129.138.12.131 and all looks well.

wcmcintosh, 2015-10-08:
Diode continues to run fine as well.

jirhiker, 2015-10-08:
If you want to set the sensitivity_multiplier=1.0 you certainly can.

here is an air script as an example
```
'''
sensitivity_multiplier: 0.5
modifier: 1
'''
def main():
    info('Jan Air Script')
    gosub('jan:WaitForMiniboneAccess')
    gosub('jan:PrepareForAirShot')
    gosub('jan:EvacPipette2')
    gosub('common:ExpandPipette2')
    gosub('common:FillPipette2')
    gosub('jan:PrepareForAirShotExpansion')
    gosub('common:ExpandPipette2')
    close(description='Outer Pipette 2')
```

follow that format exactly. the space is essential
```
sensitivity_multiplier:<space><value>
```

closed 2015-10-10 by jirhiker