NMGRL/pychron #683: pychron startup - loads analyses

wcmcintosh, 2015-10-20:
Minor point:

When pychron starts up, it loads some previous analyses, but doesn't show them in the browser

This causes no problems, but is it desirable?

jirhiker, 2015-10-20:
you'll have to elaborate. 
first no analyses are loaded on "start up", only when a task with a browser is opened. 
second a progress dialog may open when loading the labnumbers if there are more than 50. This is not loading analyses its loading the labnumbers. If no labnumbers are selected then no analyses will show up in the analyses table.

wcmcintosh, 2015-10-20:
OK not exactly startup, but after I closed the hints window, the pipeline window opened, then, without selecting a task, a popup window showed that a series of either analyses or analyses numbers was being loaded.  The popup then closed.  I have stopped and restarted and I can't get this behavior to repeat, although I have seen it before.

closed 2015-10-26 by jirhiker