NMGRL/pychron #688: error following viewing of history

wcmcintosh, 2015-10-22:
Is this the issue that I am meant ot solve myself?

    for i, x, y in zip(inds, xs, ys):
TypeError: zip argument #2 must support iteration

I think I see that xs is not satisfying zip

But I'm not sure where to go from here

```
Traceback (most recent call last):
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/enable/qt4/base_window.py", line 198, in paintEvent
    self.handler.paintEvent(event)
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/enable/qt4/base_window.py", line 50, in paintEvent
    self._enable_window._paint(event)
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/enable/abstract_window.py", line 467, in _paint
    self.component.draw(gc, view_bounds=(0, 0, size[0], size[1]))
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/enable/component.py", line 421, in draw
    self._draw(gc, view_bounds, mode)
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/enable/component.py", line 773, in _draw
    self._dispatch_draw(layer, gc, view_bounds, mode)
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/enable/container.py", line 327, in _dispatch_draw
    component._dispatch_draw(layer, gc, new_bounds, mode)
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/enable/container.py", line 327, in _dispatch_draw
    component._dispatch_draw(layer, gc, new_bounds, mode)
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/enable/component.py", line 793, in _dispatch_draw
    handler(gc, view_bounds, mode)
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/enable/component.py", line 871, in _draw_overlay
    overlay.overlay(self, gc, view_bounds, mode)
  File "/Users/mcintosh/Programming/git/pychron/pychron/graph/tools/info_inspector.py", line 96, in overlay
    lines = self.tool.assemble_lines()
  File "/Users/mcintosh/Programming/git/pychron/pychron/graph/tools/point_inspector.py", line 71, in assemble_lines
    for i, x, y in zip(inds, xs, ys):
TypeError: zip argument #2 must support iteration

```

jirhiker, 2015-10-22:
You probably wouldn't be able to resolve this yourself but you did interpret the exception correctly. 

The place to go is the source code, always. The specific line in the source code that raised the exception is provided i.e
```
File "/Users/mcintosh/Programming/git/pychron/pychron/graph/tools/point_inspector.py", line 71, in assemble_lines
    for i, x, y in zip(inds, xs, ys):
```
So I went to the point_inspector file, started at line 71 and looked at what was defining inds, xs, and ys

here is the relevant snippet
```
inds = self.get_selected_index()
lines = []
convert_index = self.convert_index
if inds is not None:
    he = hasattr(self.component, 'yerror')

    ys = comp.value.get_data()[inds]
    xs = comp.index.get_data()[inds]
    for i, x, y in zip(inds, xs, ys):            
```
for clarity im going to remove the irrelevant lines
```
inds = self.get_selected_index()
if inds is not None:
    ys = comp.value.get_data()[inds]
    xs = comp.index.get_data()[inds]
    for i, x, y in zip(inds, xs, ys):
```
The problem occurs if inds is an empty list. The if statement is only testing if its None
so the solution is to change the if statement to
```
if inds is not None and len(inds):
```
this way if inds==None or inds==[] the if statement evaluates to False

Pop Quiz. Why would the following not work
```
if len(inds) and inds is not None:
```

wcmcintosh, 2015-10-23:
Is it because len(inds) generates an error if inds is None, before checking that inds is not None?

jirhiker, 2015-10-23:
correct!

closed 2015-10-23 by jirhiker