NMGRL/pychron #694: pipeline IC factors

wcmcintosh, 2015-10-28:
IC factor tried on different data set

gave following error

```
Traceback (most recent call last):
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/pyface/ui/qt4/action/action_item.py", line 343, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 248, in run
    self._run_pipeline()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 369, in _run_pipeline
    self._run('run pipeline', 'run_pipeline', close_all=True)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 348, in _run
    if not getattr(self.engine, func)():
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 434, in run_pipeline
    node.run(state)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/find.py", line 176, in run
    if self._run_group(state, gid, list(ans)):
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/find.py", line 183, in _run_group
    refs = self.dvc.find_references(times, atype, hours=self.threshold)
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 395, in find_references
    records = self.make_analyses(records)
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 423, in make_analyses
    ret = progress_loader(records, wrapper, threshold=1)
  File "/Users/mcintosh/Programming/git/pychron/pychron/core/progress.py", line 84, in progress_loader
    return list(gen(progress))
  File "/Users/mcintosh/Programming/git/pychron/pychron/core/progress.py", line 64, in gen
    r = func(x, prog, i, n)
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 420, in <lambda>
    calculate_f_only=calculate_f_only, *args)
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 678, in _make_record
    a.set_production(pname, prod)
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc_analysis.py", line 287, in set_production
    self.production_ratios = r.to_dict(('Ca_K', 'Cl_K'))
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/meta_repo.py", line 204, in to_dict
    return {t: ufloat(getattr(self, t), getattr(self, '{}_err'.format(t))) for t in keys}
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/meta_repo.py", line 204, in <dictcomp>
    return {t: ufloat(getattr(self, t), getattr(self, '{}_err'.format(t))) for t in keys}
AttributeError: 'Production' object has no attribute 'Ca_K'

```

wcmcintosh, 2015-10-28:
Running pipeline blanks gives same error

jirhiker, 2015-10-28:
Most likely your meta repo is not up to date or there is a problem with that irradiation

easiest thing to do is delete data/.dvc/meta . Pychron will clone the meta repo automatically on startup. 

if problem persists then the Production file is: 1) missing, 2) not in the correct location or 3) incomplete.

What irradiation is causing this error.

closed 2015-10-29 by jirhiker