NMGRL/pychron #699: Diode laser position not moving

mjz1983, 2015-11-04:
Received this error message (see screen shot). After firing laser at first of three holes, message appeared on screen. Laser kept rastering in the same hole, although it was supposed to move to the second (and then third) holes for this step. This problem occurred twice. Restarted PyDiode after first crash. Stopped experiment after second crash.

[image]

jirhiker, 2015-11-04:
could not replicate error. launched pydiode and pyexperiment. ran a test analysis at low power and patterning worked as expected; executed square pattern at hole 4 then 5 then 6. 

When you ( @mjz1983 ) say crash do you mean that pydiode quit unexpectedly or simply that you received the PositionError message?

mjz1983, 2015-11-04:
Replicated the error during first analyses this morning. I received the same error (same numbers, same hole requested). PyDiode continued to move to the correct position. When I clicked OK, PyDiode quit unexpectedly.

closed 2015-11-05 by jirhiker