NMGRL/pychron #700: error on starting pipeline table

wcmcintosh, 2015-11-09:
```
Traceback (most recent call last):
  File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/traits/trait_notifiers.py", line 340, in __call__
    self.handler( *args )
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 560, in _selected_pipeline_template_changed
    self._set_template(new)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 501, in _set_template
    self.dvc)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/template.py", line 65, in render
    node = self._node_factory(klass, ni)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/template.py", line 83, in _node_factory
    node = getattr(mod, klass)()
AttributeError: 'module' object has no attribute 'TableNode'

```

jirhiker, 2015-11-09:
you have out of date template files. delete all files in setupfiles/pipeline/templates

jirhiker, 2015-11-10:
see issue #701. should resolve this issue in the future

closed 2015-11-10 by jirhiker