NMGRL/pychron #716: Spectrometer Parameters DVC

jirhiker, 2016-01-17:
finish implementing spectrometer view for DVCAnalysis

fix IsoDBTransfer so that it actually populates the spectrometer file

retroactively set spec_sha and spectrometer file for all analyses already imported

closed 2016-01-19 by jirhiker

wcmcintosh, 2016-01-19:
Hi Jake:

	spec_sha? 

Bill

William McIntosh
Geochronologist/Volcanologist
NM Bureau of Geology
Associate Professor
Earth and Environmental Science Dept
NM Tech
801 Leroy Place
Socorro, NM 87801
575-835-5271
mcintosh@nmt.edu

jirhiker, 2016-01-19:
spec_sha is a pointer for the spectrometer file which stores gains, deflections, and source parameters

every analysis has a spec_sha but most analyses are collected using the same spectrometer parameters so only one spec_sha is need. 

the sha comes from SHA-255, a hashing algorithm. the hash is generated from the contents of the file.  

here is an example of a spectrometer file 
https://github.com/NMGRLData/Irradiation-NM-278/blob/master/5ea1bf6d1c82476da752672bbe51f7b2878f52bb.json