NMGRL/pychron #799: error when attempting recall from nm-281 repository

wcmcintosh, 2016-05-06:
```
ssing                     : 2016-05-06 16:16:38,800 DEBUG     (MainThread) recalling records <pychron.database.records.isotope_record.DVCIsotopeRecordView instance at 0x11d832128>
asdfas False 24751-01
rrr 24751-01
DVC                                     : 2016-05-06 16:16:39,033 INFO      (MainThread) Analysis 24751-01 not available. Trying to clone repository "Irradiation-NM-281"
requests.packages.urllib3.connectionpool: 2016-05-06 16:16:39,417 INFO      (MainThread) Starting new HTTPS connection (1): api.github.com
requests.packages.urllib3.connectionpool: 2016-05-06 16:16:39,748 DEBUG     (MainThread) "GET /orgs/NMGRLData/repos HTTP/1.1" 401 83
root                                    : 2016-05-06 16:16:39,751 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2016-05-06 16:16:39,752 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2016-05-06 16:16:39,752 CRITICAL  (MainThread) File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/traits/trait_notifiers.py", line 520, in _dispatch_change_event
    self.dispatch( handler, *args )
root                                    : 2016-05-06 16:16:39,752 CRITICAL  (MainThread) File "/Users/mcintosh/anaconda/lib/python2.7/site-packages/traits/trait_notifiers.py", line 483, in dispatch
    handler( *args )
root                                    : 2016-05-06 16:16:39,752 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_task.py", line 370, in _handle_dclicked
    self._recall_item(new.item)
root                                    : 2016-05-06 16:16:39,752 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_task.py", line 313, in _recall_item
    self.recall(item, open_copy=open_copy)
root                                    : 2016-05-06 16:16:39,752 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_task.py", line 186, in recall
    ans = self.dvc.make_analyses(records)
root                                    : 2016-05-06 16:16:39,752 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 506, in make_analyses
    ret = progress_loader(records, func, threshold=1, step=25)
root                                    : 2016-05-06 16:16:39,752 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/core/progress.py", line 97, in progress_loader
    items = list(gen())
root                                    : 2016-05-06 16:16:39,752 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/core/progress.py", line 79, in gen
    r = func(x, prog, i, n)
root                                    : 2016-05-06 16:16:39,752 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 504, in func
    calculate_f_only=calculate_f_only, *args)
root                                    : 2016-05-06 16:16:39,752 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 949, in _make_record
    self.sync_repo(expid)
root                                    : 2016-05-06 16:16:39,752 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 584, in sync_repo
    names = self.remote_repositories()
root                                    : 2016-05-06 16:16:39,753 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 543, in remote_repositories
    ri = gi.get_repository_names(self.organization)
root                                    : 2016-05-06 16:16:39,753 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/git/hosts/__init__.py", line 84, in get_repository_names
    return [repo['name'] for repo in self.get_repos(organization)]
root                                    : 2016-05-06 16:16:39,753 CRITICAL  (MainThread) KeyError: 'name'
root                                    : 2016-05-06 16:16:39,753 CRITICAL  (MainThread) ============ End Exception ==========
Analysis Browser                        : 2016-05-06 16:16:39,753 DEBUG     (MainThread) recalling records <pychron.database.records.isotope_record.DVCIsotopeRecordView instance at 0x11d832128>

Process finished with exit code 139

```

wcmcintosh, 2016-05-06:
Above error crashed pychron

jirhiker, 2016-05-06:
could not replicate error

Did you setup github in Preferences? dvc in Preferences?

closed 2016-05-09 by jirhiker