NMGRL/pychron #848: diff

wcmcintosh, 2016-06-10:


wcmcintosh, 2016-06-10:
Diff view for unknowns now comparing bs corrected beams with mass spec, but mass spec beams are also baseline corrected

I think that 36Ar in pychron data summary is IC corrected, burt mass spec number I think is not IC corrected - this may play into diff

jirhiker, 2016-06-10:
can you please add comments to existing issues instead of always creating a new issue. This comment belongs in #844.

We need to discuss mass spec's handling of fN. I am pulling it out of PDPTable and BaselineChangeableItemsTable, but in the case of PDPTable there are only ~290 rows. So its not clear when mass spec is saving the filtered points. 

The last update to PDPTable was recent so it seems the correct place and not some old table that was never removed

wcmcintosh, 2016-06-10:
One other goofy thing about mass spec.  When you change the curve fitting preferences to 10 sigma and refit only CDD baseline, the n sigma number appears to change for all baseline isotopes.  Is this an inherent limitation of the mass spec database schema?

wcmcintosh, 2016-06-10:
I also note that for BU-FC-J-5681, the beam numbers and uncertainties and baselines and uncertainties for filtered isotopes currently all match, with the exception of 36Ar which is uncorrected in mass spec.  Hallelujah! Linear regression si not random after all!

jirhiker, 2016-06-10:
same input == same output

wcmcintosh, 2016-06-10:
same process is also required, e.g. point elimination

wcmcintosh, 2016-06-10:
also timely saving of correctly correctly values

wcmcintosh, 2016-06-10:
I see that diff is showing the correct N and fN for baselines,
but fN=N for pychron beams and for both mass spec baselines and beams.

Diff correctly shows that filtering is on for both baselines and beams
[image]

jirhiker, 2016-06-12:
- [ ] add ratios

wcmcintosh, 2016-06-13:
Diff getting a lot closer.  For 64101-01C  (see png) all bs corrected isotopes agree pretty well,, but I see some unexplained differences in "total" isotopes.  Am I correct that this "total" value is bs and bl corrected?  
Have a look at 38Ar total for 64101-01C.  Pychron value is 0.38378 and MS value is 0.386429.  MS value is indeed the "fully" corrected value from the MS summary.  Pychron summary shows same corrected value (0.386429), but this is not the pychron value shown by diff.  Relevant blank looks the same in pychron and ms.

[image]

wcmcintosh, 2016-06-13:
please add comparison of decay constants

wcmcintosh, 2016-06-14:
Diff view for blanks not working in current version

wcmcintosh, 2016-06-14:
I'm wondering if some of the previous changes in diff may have been lost along the way.  Specifically: ability to compare plan corrected isotope values (was working fine) and displaying more significant digits in the result pane after isotope fitting.  Both these features were previously working fine.

jirhiker, 2016-06-14:
Not sure what your are saying.

1. what is plain corrected isotope values? diff is doing both baseline corrected and total intensities (total having an flexible meaning). I never had comparison of baseline and blank corrected intensities

2. How many digits do you want

wcmcintosh, 2016-06-15:
Another little mystery. Tried setting 36 errors to zero in pychron and mass spec.  Pychron seems to change the values back to the original values when it feels like it.  Diff sheet attached shows that for mass spec bs corrected 36Ar agrees with pychron, as do blank values.  All 36 errors are zero as hoped.  However,  the total 36Ar for mass spec is way off.  not sure why.

[image]

wcmcintosh, 2016-06-15:
pychron manual saving now working well
ms manual saving continues to be a little flaky
still having problems with diff values for manually edited ms data, as shown in screenshot above
comparison with ms recall data looks good.  uncertainties seem to match with 36 errors set to zero
but values recovered by diff (36Ar total, 40*39, age) do not match mass spec recall
database not updated

wcmcintosh, 2016-06-15:
almost there
one more issue (see attached image)
40*39 and error agrees perfectly
but age error still different

[image]

wcmcintosh, 2016-06-16:
error on attempted diff of 64101-01C

```
call analyis va
DVC                                     : 2016-06-15 22:25:19,740 DEBUG     (MainThread) Make analysis time, total: 0.110639810562, n: 1, average: 0.110639810562
Analysis Browser                        : 2016-06-15 22:28:55,376 DEBUG     (MainThread) diff analysis
MassSpecDatabaseAdapter                 : 2016-06-15 22:28:55,379 INFO      (MainThread) using pymysql
MassSpecDatabaseAdapter                 : 2016-06-15 22:28:55,379 INFO      (MainThread) 4969061936 connecting to database mysql+pymysql://wmcintosh:Wmcintosh40*39@129.138.12.160/massspecdata?connect_timeout=5
MassSpecDatabaseAdapter                 : 2016-06-15 22:28:55,380 INFO      (MainThread) testing database connection get_database_version
MassSpecDatabaseAdapter                 : 2016-06-15 22:28:59,350 DEBUG     (MainThread) MassSpec Database Version: 16.0
MassSpecDatabaseAdapter                 : 2016-06-15 22:28:59,351 INFO      (MainThread) closing test session
MassSpecDatabaseAdapter                 : 2016-06-15 22:28:59,551 INFO      (MainThread) connected to db mysql+pymysql://wmcintosh:Wmcintosh40*39@129.138.12.160/massspecdata?connect_timeout=5
MassSpecDatabaseAdapter                 : 2016-06-15 22:29:01,601 WARNING   (MainThread) =========== Database exception =============
MassSpecDatabaseAdapter                 : 2016-06-15 22:29:01,601 WARNING   (MainThread) r
MassSpecDatabaseAdapter                 : 2016-06-15 22:29:01,610 WARNING   (MainThread)   File "/Users/mcintosh/Programming/git/pychron/pychron/mass_spec/mass_spec_recaller.py", line 51, in find_analysis
    rec.sync(dbrec)
,  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis.py", line 348, in sync
    self._sync(obj, **kw)
,  File "/Users/mcintosh/Programming/git/pychron/pychron/mass_spec/mass_spec_analysis.py", line 77, in _sync
    self.r
,  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/isotope_group.py", line 343, in __getattr__
    raise AttributeError(attr)

root                                    : 2016-06-15 22:29:02,119 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2016-06-15 22:29:02,119 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2016-06-15 22:29:02,119 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/pyface/ui/qt4/action/action_item.py", line 343, in _qt4_on_triggered
    self.controller.perform(action, action_event)
root                                    : 2016-06-15 22:29:02,120 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
root                                    : 2016-06-15 22:29:02,120 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
root                                    : 2016-06-15 22:29:02,120 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/browser_task.py", line 179, in diff_analysis
    if editor.setup(left):
root                                    : 2016-06-15 22:29:02,120 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/editors/diff_editor.py", line 161, in setup
    right = self._find_right(left)
root                                    : 2016-06-15 22:29:02,120 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/editors/diff_editor.py", line 192, in _find_right
    left.step)
root                                    : 2016-06-15 22:29:02,120 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/mass_spec/mass_spec_recaller.py", line 51, in find_analysis
    rec.sync(dbrec)
root                                    : 2016-06-15 22:29:02,120 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis.py", line 348, in sync
    self._sync(obj, **kw)
root                                    : 2016-06-15 22:29:02,121 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/mass_spec/mass_spec_analysis.py", line 77, in _sync
    self.r
root                                    : 2016-06-15 22:29:02,121 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/processing/isotope_group.py", line 343, in __getattr__
    raise AttributeError(attr)
root                                    : 2016-06-15 22:29:02,121 CRITICAL  (MainThread) AttributeError: r
root                                    : 2016-06-15 22:29:02,121 CRITICAL  (MainThread) ============ End Exception ==========
2016-06-15 22:29:43.224 python[11596:1016890] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
root                                    : 2016-06-15 22:29:43,256 CRITICAL  (Dummy-1   ) ============ Exception ==============
root                                    : 2016-06-15 22:29:43,256 CRITICAL  (Dummy-1   ) Traceback (most recent call last):
root                                    : 2016-06-15 22:29:43,256 CRITICAL  (Dummy-1   ) File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/traits/trait_notifiers.py", line 520, in _dispatch_change_event
[... cut ...]
root                                    : 2016-06-15 22:29:43,257 CRITICAL  (Dummy-1   ) File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/pyface/ui/qt4/action/action_item.py", line 201, in _on_action_enabled_changed
    self.control.setEnabled(action.enabled)
root                                    : 2016-06-15 22:29:43,257 CRITICAL  (Dummy-1   ) RuntimeError: Internal C++ object (PySide.QtGui.QAction) already deleted.
root                                    : 2016-06-15 22:29:43,257 CRITICAL  (Dummy-1   ) ============ End Exception ==========

```

closed 2016-08-06 by jirhiker