NMGRL/pychron #852: error on diff view after successfully recalling bu-FC-J-5681

wcmcintosh, 2016-06-12:
active branch=release/v2.5

active analyses=bu-FC-J-5681

```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/pyface/ui/qt4/action/action_item.py", line 343, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/mcintosh/.pychron/updates/pychron/pipeline/tasks/browser_task.py", line 171, in diff_analysis
    if editor.setup(left):
  File "/Users/mcintosh/.pychron/updates/pychron/pipeline/editors/diff_editor.py", line 164, in setup
    right = self._find_right(left)
  File "/Users/mcintosh/.pychron/updates/pychron/pipeline/editors/diff_editor.py", line 195, in _find_right
    left.step)
  File "/Users/mcintosh/.pychron/updates/pychron/mass_spec/mass_spec_recaller.py", line 68, in find_analysis
    prefs = db.get_latest_preferences(iso.IsotopeID)
  File "/Users/mcintosh/.pychron/updates/pychron/mass_spec/database/massspec_database_adapter.py", line 157, in get_latest_preferences
    q = q.join(IsotopeTable)
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 1946, in join
    from_joinpoint=from_joinpoint)
  File "<string>", line 2, in _join
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/sqlalchemy/orm/base.py", line 201, in generate
    fn(self, *args[1:], **kw)
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2089, in _join
    outerjoin, create_aliases, prop)
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2157, in _join_left_to_right
    self._join_to_left(l_info, left, right, onclause, outerjoin)
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2261, in _join_to_left
    "Tried joining to %s, but got: %s" % (right, ae))
InvalidRequestError: Could not find a FROM clause to join from.  Tried joining to <class 'pychron.mass_spec.database.massspec_orm.IsotopeTable'>, but got: Can't find any foreign key relationships between 'Join object on Join object on PreferencesTable(4803396624) and AnalysesChangeableItemsTable(4788538000)(4900340176) and DataReductionSessionTable(4802736208)' and 'IsotopeTable'.

```

wcmcintosh, 2016-06-12:
diff view gives same error for unknown data

wcmcintosh, 2016-06-12:
request - could diff view please include "fully" corrected isotope values to compare with mass spec summary values

something funny is going on with 36Ar values, probably related to timing of IC correction

I note that while isotope values in pychron and mass spec (for test sample 64101-01C anyway) are getting quite close, the age results from the two programs remain >1% different

having the corrected isotope values may help sort out this problem.  So far I haven't been able to entirely close the loop, especially for 36Ar.

wcmcintosh, 2016-06-12:
diff has definitely been working better with the recent rounds of changes, except of the last batch of changes which apparently has a bug.  Including some ratios in diff, particularly 40*/39 might also be useful.  Pychron and mass spec currently both include 40/36 and 40/38 ratios.  40/38 ratios agree between programs. 40/36 do not, partially or entirely related to IC correction.  Could this also explain age differences between pychron and mass spec?

wcmcintosh, 2016-06-12:
note that this new issue was automatically created when the automatic bug report was submitted.  I would have preferred to add it to an existing issue, even though I think that the formerly appropriate issue has been closed.

jirhiker, 2016-06-12:
related to issue #848

jirhiker, 2016-06-12:
added fully corrected but not sure if mass spec is recording interference corrected values or just 
bs,bk, and ic.

closed 2016-06-13 by jirhiker