NMGRL/pychron #912: Possible blank fit problem?

wcmcintosh, 2016-08-17:
Jake, I am fitting blanks for NM281C monitors, and wonder about a couple of things.

1) previously, could one hover over a blank in the fit screen and see the number analysis number pop up?
2) in the blank options box, how can one delete or null out an isotope line?
3)something looks funny in the reference display for some isotopes. 40 and 36 look ok
but 41 and 37 have identical values, and 38 and 39 have identical values

wcmcintosh, 2016-08-17:
Actually, not the exactly same values, but the reference isotope errors don't seem right - they don't jive with the saved values

jirhiker, 2016-08-17:
1. functionality moved to `Inspector` pane
2. click the `Enabled` checkbox
3. ?

jirhiker, 2016-08-17:
this is not the correct place for such discussion. Use Slack for questions and discussion

closed 2016-08-17 by jirhiker