NMGRL/pychron #914: flux fitting request

wcmcintosh, 2016-08-18:
from the flux fitting pane, would it be possible when points are tagged as invalid, to eliminate the point and rescale the plot? Current version requires restarting pipeline after marking invalids to fix scale. Points  marked as invalid remain as valid points on plot.

Not high priority, can work around.

wcmcintosh, 2016-08-18:
Another note on Flux.  Traditionally we have saved the  actual mean J values for monitors and predicted values for unknowns.  Is this an option in Pychron? If I uncheck save, monitor values remain zero.  Should there be a column for saving actual mean monitor values,  as an alternative to predicted values?  Should both be saved?

jirhiker, 2016-08-18:
added mean j to flux file. currently no way of viewing the saved mean j without looking at the file.

closed 2018-05-11 by jirhiker