NMGRL/pychron #938: Jan experiment que - error after first analysis complete

nmgrluser, 2016-08-31:
active branch=None

active analyses=bu-FC-J-6814

```
Traceback (most recent call last):
  File "/Users/jan/Programming/git/pychron/pychron/core/helpers/exception_helper.py", line 275, in run_with_except_hook
    run_old(*args, **kw)
  File "/Users/jan/anaconda/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/Users/jan/Programming/git/pychron/pychron/experiment/experiment_executor.py", line 450, in _execute
    self._execute_queue(i, exp)
  File "/Users/jan/Programming/git/pychron/pychron/experiment/experiment_executor.py", line 571, in _execute_queue
    self._join_run(spec, run)
  File "/Users/jan/Programming/git/pychron/pychron/experiment/experiment_executor.py", line 666, in _join_run
    self._do_run(run)
  File "/Users/jan/Programming/git/pychron/pychron/experiment/experiment_executor.py", line 734, in _do_run
    run.save()
  File "/Users/jan/Programming/git/pychron/pychron/experiment/automated_run/automated_run.py", line 1037, in save
    self._persister_save_action('post_measurement_save')
  File "/Users/jan/Programming/git/pychron/pychron/experiment/automated_run/automated_run.py", line 1496, in _persister_save_action
    getattr(self.dvc_persister, func)(*args, **kw)
  File "/Users/jan/Programming/git/pychron/pychron/dvc/dvc_persister.py", line 209, in post_measurement_save
    self.active_repository.add(p, commit=False, msg_prefix=msg_prefix)
  File "/Users/jan/Programming/git/pychron/pychron/git_archive/repo_manager.py", line 692, in add
    self._add_to_repo(dest, msg, **kw)
  File "/Users/jan/Programming/git/pychron/pychron/git_archive/repo_manager.py", line 759, in _add_to_repo
    index.add(p)
  File "/Users/jan/anaconda/lib/python2.7/site-packages/git/index/base.py", line 777, in add
    self.entries[(entry.path, 0)] = IndexEntry.from_base(entry)
  File "/Users/jan/anaconda/lib/python2.7/site-packages/gitdb/util.py", line 237, in __getattr__
    self._set_cache_(attr)
  File "/Users/jan/anaconda/lib/python2.7/site-packages/git/index/base.py", line 122, in _set_cache_
    fd = lfd.open(write=False, stream=False)
  File "/Users/jan/anaconda/lib/python2.7/site-packages/gitdb/util.py", line 314, in open
    raise IOError("Lock at %r could not be obtained" % self._lockfilepath())
IOError: Lock at '/Users/jan/Pychron/data/.dvc/repositories/Heizler/.git/index.lock' could not be obtained

```

wcmcintosh, 2016-08-31:
logger window contents after last quit

'''
AutomatedRun                             -- No value "sensitivity_multiplier" in metadata
ClientExtractionLineManager              -- Invalid valve name=None, description=CO2 Laser to Felix
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
PeakCenter                               -- Peak center failed to move to a baseline position
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
AutomatedRunPersister                    -- No database instance. Not saving post measurement to isotopedb database
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
[... cut ...]
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25
SampleBrowserModel                       -- Set "RECENT (hrs)" in Preferences.
"RECENT (hrs)" is located in the "Browser" category
ClientSwitchManager                      -- Valve word length is too short. All valve states will not be updated! Word:5, Num Valves: 25

'''

wcmcintosh, 2016-08-31:
it appears that valve has multiple names

CO2 to Obama (Pychron computer)
CO2 to Felix (Jan computer)
CO2 to Bone (Felix Mac)

wcmcintosh, 2016-08-31:
First analysis in queue ran and line is green, but not saved

jirhiker, 2016-08-31:
same issue as #917. 

Warning message 
```ClientExtractionLineManager -- Invalid valve name=None, description=CO2 Laser to Felix```
not relevant

closed 2016-08-31 by jirhiker