NMGRL/pychron #965: CO2 Commuter apparent failure

nmgrluser, 2016-09-25:
Was working fine
Had an issues connecting to the chiller when trying to fire the CO2 laser
Chiller was on and seemed to be working fine
Changed script in initialization.xml to chiller- false
CO2 computer went crazy with and the video was choppy. Like when a  window skips when you drag it but for the whole screen. The only thing that was clear was the CO2 laser camera
Hard restart on the CO2 computer and then no longer responsive
-could not acces through screen share
-could not ping the computer
-could not plug in hdml cable and make it work
Tried resetting sever other time including unplugging it and still wont come back online

Bill and I (Nels) worked on this for a while via phone

wcmcintosh, 2016-09-25:
still no ping

wcmcintosh, 2016-09-25:
jake- can you send us info on setting up another co2 mini from backup?

jirhiker, 2016-09-25:
the trick is
1. unplug the co2 usb cable from the front of the computer
2. restart

closed 2016-09-26 by wcmcintosh