NMGRL/pychron #971: trouble with production ratios

wcmcintosh, 2016-09-26:
When trying to make a new production ratio file for cocktail, can't make new production file the than triga.json, trig.json.json, etc. than these not recognized

jirhiker, 2016-09-27:
Definitely do not edit the .dvc/MetaData repository if you are not comfortable/familiar with git. 

You should not be editing/interacting with irradiations, other than using the most basic features (ones that you've been instructed on how to use)

closed 2016-09-27 by jirhiker

wcmcintosh, 2016-09-27:
Jake:

One of the most basic features is changing production ratios, but I am not able to change to a different production ratio file for the cocktail irradiation.

Bill

jirhiker, 2016-09-28:
Productions ratios have been a huge issue and my solution has not been finalized or fully tested. Nor has production ratio editing been a priority up until now. 

My response was testy only because I do not want users mucking around with the .dvc directory directly for fear of causing rather annoying git related issues. Directly editing the .dvc takes are real clear understanding of how DVC operates, something I have no expectation that users should have at this point. 

In general Im unhappy with much of how "cocktails" have been handled by pychron and I think a redesign is in order. 

I apologize for the tone of my original response.

wcmcintosh, 2016-09-30:
Jake:

	I shoud have done more testing of cocktail and air related procedures earlier. 

	Currently there are a few different  issues with cocktails on Pychron, production ratios and J values.

	Hope all is well at the workshop.

Bill