NMGRL/pychron #979: Oddball CO2 stage behavior

nmgrluser, 2016-09-30:
Haven't seen this one before.  When CO2 hole n is requested, stage is in most (but not all) cases going to n + 1, and identifying the target hole correctly as n + 1.  Currently recalibrating, but previous semi-auto calibration seemed fine.

Ideas?   The behavior was the same regardless of whether pyexperiment or the user was requesting the hole move.

Bill

Note: Restarting = no change.  Recalibrating = all better.

Something to watch out for.

jirhiker, 2016-10-01:
sounds like the first auto calibration identified hole 2 as hole 1 etc...

wcmcintosh, 2016-10-04:
Jake:

	When hole  n was requested, Pychron was going to hole n+1 , but it identified it corrected as as n+1, not n.  Would this be the case with an offset calibration?

Bill

jirhiker, 2016-10-04:
how do you know it identified it as n+1?

take the following scenario.
You do an semi-auto tray calibration and hole2 is identified as hole1. So for hole1 pychron stores the current coordinates, xA,yA, (which are actually the coordinates for hole2) as the correct position.

then when you then ask pychron to go to hole1, it first checks if the hole has a corrected position. In this case it does (xA,yA) so pychron goes to xA,yA then does one auto positioning iteration at xA,yA. 

However since xA,yA is in reality the position for hole2 it would appear that when you enter holeN you get holeN+1

wcmcintosh, 2016-10-05:
Hi Jake:

	My basis for saying that Pychron identified the hole as n+1 is that the number displayed by the hole on the video after the move was n+1.  Perhaps Pychron uses different criteria for identifying the targeted hole.

Bill

jirhiker, 2016-10-05:
The hole number displayed on video is dynamically determined. pychron takes the current x,y and compares that to the nominal calibrated positions for all holes looking for a match. 

I still think the problem was hole2 was identified as hole 1 during semi-auto calibration

closed 2016-10-24 by jirhiker