NMGRL/pychron #984: CO2 problem

wcmcintosh, 2016-10-10:
I think that Matt's problem was due to USB cable being disconnected from back of laser controller.  Discovered this after power cycling everything.  Now Python boots up fine but gives repeated g_dimension exception after view CO2 laser

wcmcintosh, 2016-10-10:
commented out g_dimension lines

seems to be running OK, unless 221-small hole file selected 

221-hole and 221-hole.center seem OK, although both don't work unless "g_dimension" lines commented out

jirhiker, 2016-10-12:
issues #980 to #984 red herrings to the real problem. 

A corrupted correction file was preventing stage_manager.stage_map from being created.

will add better warning and exception handling to this part of pychron

closed 2016-10-18 by jirhiker