NMGRL/pychron #987: Experiment Queue pause problem

wcmcintosh, 2016-10-18:
Can insert Pause, looks for labnumber, pulldown empty

jirhiker, 2016-10-18:
is this on jan or felix

wcmcintosh, 2016-10-18:
jan

wcmcintosh, 2016-10-18:
On Jan Experiment queue writer, I see that the available lab numbers for blanks, airs, etc. include felix lab numbers as well as Jan lab numbers.  Should they be Jan only, once Jan is selected in the queue pane?

jirhiker, 2016-10-18:
"lab numbers for blanks, airs, etc. include felix lab numbers as well as Jan lab numbers" is that because you have "Unirradiated" or "NoIrradiation" selected for the irradiation?  The irradiation drop down should be unselected,(default,---,Irradiation) when selecting from "Special Labnumbers"?

jirhiker, 2016-10-18:
Well what i just said may not be the case. But i consider the labnumber drop down irrelevant when selecting "special labnumbers". 

you can run a pause with any labnumber. just make sure you set the extraction script to "pause" and measurement script to null.

closed 2017-10-19 by jirhiker