NMGRL/pychron #1005: browser for flux calculation

wcmcintosh, 2016-10-25:
new option to use browser for flux calculation selection working well

points marked as invalid not being excluded from plot after marking  (or even after deleting them from the analysis list

jirhiker, 2016-10-25:
which plot

wcmcintosh, 2016-10-25:
The fit flux plot

jirhiker, 2016-10-25:
"points marked as invalid not being excluded from plot after marking (or even after deleting them from the analysis list"

Can not replicate this behavior. 1. clicking on point/selected a set of points sets the tag to "omit" using "Set Invalid" is not necessary. 2. deleting analyses from the analysis pane causes flux recalc and the graph to update.

wcmcintosh, 2016-10-28:
During fit flux editing, is there a way I can set bad analyses to invalid?

jirhiker, 2016-10-28:
the same ways you set analyses as invalid in an ideogram don't work?
1. select the analyses in the Analyses pane, then use toolbar action "Set Tag", or "Set Invalid"
2. select analyses graphically,                     "                                                                               "
3. right-click analysis, set invalid

wcmcintosh, 2016-10-28:
Jake:

	When I wrote the ticket, the problem was that after setting analyses invalid, the invalid analyses were not disappearing and the axes were not resizing.

	Omit was working fine, invalid marking was working fine, invalid removal and auto graph replotting was not working.  

	I think that this replot is working fine in the current version, although there appear to be some tagging glitches, ticket submitted.

Bill

closed 2019-07-21 by jirhiker