NMGRL/pychron #1006: Pychron Config Manager

jirhiker, 2016-10-27:
add something akin to conda but instead of python dependencies download pychron configuration files. 

options to download individual file or directory trees. 
if individual file and -e option used user prompted for information

this should be a separate project from pychron. make a fork of conda and just start hacking away at it. 
conda-pychron

jirhiker, 2016-10-27:
new project name is `chronda` !!!

jirhiker, 2016-10-27:
use conda to install packages and anaconda.org to host.

use chronda to create new pychron environments from downloaded packages. 

`chronda create -n Pychron_dev development` == create pychron env Pychron_dev using the development package.

wcmcintosh, 2016-10-28:
Nice.

closed 2022-04-15 by jirhiker