NMGRL/pychron #1027: Valve status, as discussed

wcmcintosh, 2016-12-02:
I'm backing up yesterdays discussion on workarounds for the valve status issue, that is: errors generated as a downstream consequence of Pyvalve's status memory leak.  Possible solutions included broadcasting instead of responding to requests, in part to reduce the need to create and destroy objects.  Other ideas include enrichening the status word, perhaps to include new elements of status such as:
 1) valve request in progress for this valve
 2) state of health indicators (vacuum, air pressure, room temp, etc. )which Pyexperiment could use to choose to abort a run

jirhiker, 2016-12-05:
1. not likely to implement
2. can be handled by AutomatedRunMonitor

broadcast would not add any benefit

jirhiker, 2016-12-05:
a working solution for valve states.
1. make sure status monitor is running on pyvalve and pyexperiment
2. the state_word then represents the in-memory states but the in-memory states should stay accurate because of the status monitor.

maybe as some thread-safe measures tho might not be necessary

closed 2019-03-17 by jirhiker