NMGRL/pychron #1030: Furnace loader OK

wcmcintosh, 2016-12-02:
Felix and Jan running well.   Furnace sample dumped perfectly with electromagnets.  Perhaps Matt did have the valves shut when he tried it. All controls working well except "Dump" button, which  is seemingly not doing anything. All other button elements fine.  Should dump button actuate valves and shutters, then dump? Or does it actuate the magnets and mark the position as dumped?  Perhaps the button should be labelled "Load" no dump, and the "Magnets" button should be labelled "Dump".  I will have a look at whether all buckets will now dump.

Could funnel position be encoded into the status word?  These would allow easy animation of the funnel graphic.

nmgrluser, 2016-12-02:
Request - Could we add a 3 second delay before initiating the dump sequence?

jirhiker, 2016-12-05:
the dump button starts execution of the dump sequence script. 
the nominal sequence is

        1. open gate valve
        2. open shutters
        3. lower funnel
        4. actuate magnets
        5. raise funnel
        6. close shutters
        7. close gate valve

available commands for the dump sequence file are
`open, close, sleep, lower_funnel, raise_funnel, actuate_magnets`

jirhiker, 2016-12-05:
`Could funnel position be encoded into the status word? These would allow easy animation of the funnel graphic.` 

getting the funnel position is not an issue so adding to the status word has no relationship or benefit to animating funnel graphic. 

Animating funnel graphic requires changing the funnel SceneCanvas item into an object the has dynamic/animatable properties. This is not necessarily easily doable especially without unintended performance consequences.

closed 2016-12-31 by jirhiker