NMGRL/pychron #1032: error on selecting isotope evolution from menu

wcmcintosh, 2016-12-03:
active branch=release/v2.5

active analyses=65194-01F,65194-01G,65194-01H,65194-01I,65194-01J

```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/traits/trait_notifiers.py", line 340, in __call__
    self.handler( *args )
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 688, in _selected_pipeline_template_changed
    self._set_template(new)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 583, in _set_template
    self.dvc)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/template.py", line 57, in render
    nodes = yd['nodes']
TypeError: list indices must be integers, not str

```

jirhiker, 2016-12-04:
the pipeline template you are selecting is corrupted

closed 2016-12-04 by jirhiker

wcmcintosh, 2016-12-04:
Can you provide info on how to correct?