NMGRL/pychron #1033: Sorcerer's apprentice

wcmcintosh, 2016-12-03:
I am running some data in the pipeline. On way or another I got a bad previous list of samples that generate exception errors.  If I could start afresh I would be OK, but the same previous list of analyses keeps loading and causing errors.  Even a Pychron restart won't clear the list.  It keeps getting loaded and greeting errors each time I run any pipeline, before I can select new samples.

How to escape???

wcmcintosh, 2016-12-03:
Overtime I try to get to the browser, the bad list loads.  How to delete the l\previous selection list.

wcmcintosh, 2016-12-04:
I solved the problem by throwing out files in pychron_dev .hidden file

jirhiker, 2016-12-04:
in the new release the relevant file to delete is

`<PychronEnvironment>/.appdata/analysis_sets`
where `<PychronEnvironment>` typically is `~/Pychron` or more explicitly in @wcmcintosh case `/Users/mcintosh/Pychron/.appdata/analysis_sets`

wcmcintosh, 2016-12-04:
Deletion of this file did free me from the loop.
Would it be a good idea for the clear button to suppress loading of previous data?

closed 2016-12-05 by jirhiker