NMGRL/pychron #1036: pychron tag troubles

wcmcintosh, 2016-12-04:
tags are having persistence problems.  Reopening a tagged data set lists some tags as bracketed strings and don't treat them correctly, e.g. excluding data tagged as invalid and omitting data tagged as omit

jirhiker, 2016-12-04:
i need to know explicitly what you are doing with tags and what datasets you are using. sounds like a combination of mismatched program versions and data versions.

wcmcintosh, 2016-12-04:
I was tagging analyses with omit and invalid interactively from the ideogram pipeline
Note the crazy tags displayed - we have seen them before
I also note that no tags are displayed from the browser list
It would be great to somehow simplify an unify the work of tagging.

[image]

jirhiker, 2016-12-04:
Tagging has been unified for quite some time. However, every time I introduce a new feature you say "no I want to be able to do it like this, and this and from this screen at this point and in a different way from another screen".  

The information you provided is not what im looking for. I want to know exactly the sequence of steps you are doing. In what circumstances does tagging work and when does it. How are you selecting the analyses for tagging. I want to know exactly the files you are working on. not the analyses. because your files are different from my files. If you don't share your changes then I can't not see them.

jirhiker, 2016-12-04:
issue resolved in develop or upcoming release

closed 2016-12-04 by jirhiker