NMGRL/pychron #1041: Isotopic Evolution pipeline tweak requests - non-urgent

wcmcintosh, 2016-12-10:
1) Could "plus" button act as "save-as" for the onscreen selections?  Current "plus" sign adds an un-populated set of selections 

2) Global pulldown for error at top of screen has different choices )SD,SEM,SEM if ...) than within-row pull down (SD,SEM,Cl)

3) Could detector baseline choice and pipeline display tab of same be cleaned up in terms of detector order (H2,H1,AX,L1, L2(CDD)) and to eliminate extra blank lines in pipeline display tab.

wcmcintosh, 2016-12-10:
Is "restore factory defaults" enabled?

wcmcintosh, 2016-12-10:
Returning to an old theme, have you thought anymore about our discussed idea of flagging evolutions that don't meet selected criteria?

jirhiker, 2016-12-10:
1. OK
2. OK
3. Don't understand

Factory defaults are enabled, the problem is the factory has not defined all the defaults. Should work for Ideogram?

WRT flagging evolutions 1) during analysis save pychron uses AI via an IsotopeClassifier to "classify", "label", "identify",etc an isotopic regression as Good or Bad with an associated probability. Classification could be extended to more "classes" e.g. Excellent, Good, Average, Bad, Very Bad,etc.
2) during isotope refit, at the Fit IsoEvo window select Iso Evo from the LHS and you will see "Goodness Threshold". If the % error on the intercept is greater than "Goodness Threshold" the regression is flagged. This is very beta and primitive and Im aware of its limitations.

wcmcintosh, 2016-12-12:
Thanks for #1 and #2.  I'll test them.
Regarding item 3:  I am requesting that the baseline fits be listed in high-to-low mass order, and that blank lines between baseline fit lines  be removed from display tab after getting isotopes.

I'll give the

wcmcintosh, 2016-12-12:
Just to clarifying: Have items 1 and 2 been addressed?  Or are the on the list?

closed 2017-04-19 by jirhiker