NMGRL/pychron #1044: chained pipelines - data reduction protocol

wcmcintosh, 2016-12-10:
This ticket is to formalize recently discussed idea of chaining pipelines to create a data reduction protocols

One thing that might be needed to implement this are modifications to start or end of pipelines to facilitate chaining.

Your concern was a user's ability to build protocols - this could be as easy as a gui to list a sequence of existing pipelines, adding additional review nodes if needed.

One option would be to not do processes (e.g.  refit isotopes) if already done. Or is pychron already fully doing this?  We had discussed whether a request to repeat a refit should be recorded in history.  I'm not sure about the outcome of this discussion.

jirhiker, 2016-12-10:
Pipeline Templates can be built by right clicking on a Node and select "Add..." e.g. Add Ideogram, Add Review. 

Current deficiencies are 1) all required"Add"s are not available, e.g. Add Blanks etc, and 2) there is no mechanism to save templates.

both of these are easily fixable. 

pychron will refit the isotopes but will not save the results if they do not differ from what's stored in the repository.  Pychron could compare the current fits with the new requested fits and any that are the same could be skipped. This seems to me a tertiary feature with little practical benefit.

wcmcintosh, 2016-12-12:
tried chain of iso ev and ideo, not working yet

wcmcintosh, 2016-12-14:
Pipeline thoughts for Jake:

Currently we have two tiers of pipeline procedures:  nodes and groups of nodes

Ideally chaining allows sequencing of groups of nodes.   

Does current version allow grouped groups of nodes? ("protocols" perhaps?)

I note that current nomenclature in some cases uses the same term (e.g. "ideogram") to reference both an individual node and the grouped node that includes it.  Maybe this is fine.

closed 2018-01-20 by jirhiker