NMGRL/pychron #1045: peak hop - recall

wcmcintosh, 2016-12-10:
When I recall ic-01-F-12, only one isotope is shown: 40Ar on L1, so other peak-hopped measurements are missing.

As fas as I can tell, isotopic fits work perfectly on 40Ar in L1.

I note that ic analyses do not even come up on Mass Spec's chronological Felix analyses list - which is probably as it should be. 

Related question:  When ic data display/recall is ironed out, what are options for exporting data to Excel or CSV format?  Matt H would use this feature.

jirhiker, 2016-12-10:
As I mentioned, pychron is currently not saving the ic data correctly and the analyses have to be retroactively fixed in order to view.  **only analyses ic-01-F-01 to ic-01-F-11 have been fixed**

IC analyses are not saved to MassSpec. 

Options for exporting are limitless. What do you want **SPECIFICALLY** and How do you want the user to interact with the export function?

wcmcintosh, 2016-12-12:
minimal requirements for CSV file
1) run number
2) mass spec
3) for each detector intercept and error

wcmcintosh, 2016-12-12:
recall of peak hop data  seems to work fine

wcmcintosh, 2016-12-12:
Pychron is updated, but recall last-run peak hop only showing one isotope -

closed 2016-12-31 by jirhiker