NMGRL/pychron #1046: plotting anything versus anything

wcmcintosh, 2016-12-10:
We have previously discussed plotting anything versus anything. I forgot is there is already a ticket on this.

Matt H would dearly love to plot a time series of cocktail data that includes Radiogenic yield, Age, and centering position. (or other factors that might affect mass spectrometer drift).

How far out is pychron from having this ability?  Beg deal or small deal?

Thanks.

jirhiker, 2016-12-10:
That is not plotting anything verse anything, that is plotting a time series. Pychron can already plot timeseries. 

Pychron also already has XY Scatter plots which truly allows for anything verse anything plots

jirhiker, 2016-12-10:
[cocktailtimeseries.pdf](https://github.com/NMGRL/pychron/files/644058/cocktailtimeseries.pdf)

jirhiker, 2016-12-10:
[image]

wcmcintosh, 2016-12-12:
time series does indeed seem to work perfectly here.  Trying X vs Y

wcmcintosh, 2016-12-12:
Is x vs Y scatter plotting enabled?

jirhiker, 2016-12-12:
There is a Pipeline Template called XYScatter

closed 2016-12-31 by jirhiker