NMGRL/pychron #1051: error plotting series of felix cocktails

wcmcintosh, 2016-12-14:
```
root                                    : 2016-12-14 11:01:51,841 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2016-12-14 11:01:51,841 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2016-12-14 11:01:51,841 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/qt4/base_window.py", line 184, in paintEvent
    self.handler.paintEvent(event)
root                                    : 2016-12-14 11:01:51,841 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/qt4/base_window.py", line 50, in paintEvent
    self._enable_window._paint(event)
root                                    : 2016-12-14 11:01:51,842 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/abstract_window.py", line 420, in _paint
    self.component.draw(gc, view_bounds=(0, 0, size[0], size[1]))
root                                    : 2016-12-14 11:01:51,842 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/component.py", line 421, in draw
    self._draw(gc, view_bounds, mode)
root                                    : 2016-12-14 11:01:51,842 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/component.py", line 763, in _draw
    self._dispatch_draw(layer, bb, view_bounds, mode)
root                                    : 2016-12-14 11:01:51,842 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/container.py", line 327, in _dispatch_draw
    component._dispatch_draw(layer, gc, new_bounds, mode)
root                                    : 2016-12-14 11:01:51,842 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/container.py", line 327, in _dispatch_draw
    component._dispatch_draw(layer, gc, new_bounds, mode)
root                                    : 2016-12-14 11:01:51,842 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/container.py", line 327, in _dispatch_draw
    component._dispatch_draw(layer, gc, new_bounds, mode)
root                                    : 2016-12-14 11:01:51,842 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/component.py", line 793, in _dispatch_draw
    handler(gc, view_bounds, mode)
root                                    : 2016-12-14 11:01:51,842 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/component.py", line 881, in _draw_underlay
    underlay.overlay(self, gc, view_bounds, mode)
root                                    : 2016-12-14 11:01:51,843 CRITICAL  (MainThread) File "/Users/argonlab2/Programming/git/pychron/pychron/graph/error_envelope_overlay.py", line 77, in overlay
    LinePlot._render_normal(gc, upts, '')
root                                    : 2016-12-14 11:01:51,843 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/chaco/lineplot.py", line 393, in _render_normal
    gc.lines(ary)
root                                    : 2016-12-14 11:01:51,843 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/kiva/agg/agg.py", line 1012, in lines
    def lines(self, *args): return _agg.GraphicsContextArray_lines(self, *args)
root                                    : 2016-12-14 11:01:51,843 CRITICAL  (MainThread) TypeError: Cannot cast array data from dtype('O') to dtype('float64') according to the rule 'safe'
root                                    : 2016-12-14 11:01:51,843 CRITICAL  (MainThread) ============ End Exception ==========
root                                    : 2016-12-14 11:02:05,694 CRITICAL  (MainThread) ============ Exception ==============
ignore exception
root                                    : 2016-12-14 11:02:05,694 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2016-12-14 11:02:05,694 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/qt4/base_window.py", line 184, in paintEvent
    self.handler.paintEvent(event)
root                                    : 2016-12-14 11:02:05,694 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/qt4/base_window.py", line 50, in paintEvent
    self._enable_window._paint(event)
root                                    : 2016-12-14 11:02:05,694 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/abstract_window.py", line 420, in _paint
    self.component.draw(gc, view_bounds=(0, 0, size[0], size[1]))
root                                    : 2016-12-14 11:02:05,695 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/component.py", line 421, in draw
    self._draw(gc, view_bounds, mode)
root                                    : 2016-12-14 11:02:05,695 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/component.py", line 763, in _draw
    self._dispatch_draw(layer, bb, view_bounds, mode)
root                                    : 2016-12-14 11:02:05,695 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/container.py", line 327, in _dispatch_draw
    component._dispatch_draw(layer, gc, new_bounds, mode)
root                                    : 2016-12-14 11:02:05,695 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/container.py", line 327, in _dispatch_draw
    component._dispatch_draw(layer, gc, new_bounds, mode)
root                                    : 2016-12-14 11:02:05,695 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/container.py", line 327, in _dispatch_draw
    component._dispatch_draw(layer, gc, new_bounds, mode)
root                                    : 2016-12-14 11:02:05,695 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/component.py", line 793, in _dispatch_draw
    handler(gc, view_bounds, mode)
root                                    : 2016-12-14 11:02:05,695 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/component.py", line 881, in _draw_underlay
    underlay.overlay(self, gc, view_bounds, mode)
root                                    : 2016-12-14 11:02:05,695 CRITICAL  (MainThread) File "/Users/argonlab2/Programming/git/pychron/pychron/graph/error_envelope_overlay.py", line 77, in overlay
    LinePlot._render_normal(gc, upts, '')
root                                    : 2016-12-14 11:02:05,695 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/chaco/lineplot.py", line 393, in _render_normal
    gc.lines(ary)
root                                    : 2016-12-14 11:02:05,695 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/kiva/agg/agg.py", line 1012, in lines
    def lines(self, *args): return _agg.GraphicsContextArray_lines(self, *args)
root                                    : 2016-12-14 11:02:05,695 CRITICAL  (MainThread) TypeError: Cannot cast array data from dtype('O') to dtype('float64') according to the rule 'safe'
root                                    : 2016-12-14 11:02:05,695 CRITICAL  (MainThread) ============ End Exception ==========
root                                    : 2016-12-14 11:02:11,389 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2016-12-14 11:02:11,389 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2016-12-14 11:02:11,389 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/qt4/base_window.py", line 184, in paintEvent
    self.handler.paintEvent(event)
root                                    : 2016-12-14 11:02:11,389 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/qt4/base_window.py", line 50, in paintEvent
    self._enable_window._paint(event)
root                                    : 2016-12-14 11:02:11,389 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/enable/abstract_window.py", line 420, in _paint
[... cut ...]
root                                    : 2016-12-14 11:02:11,390 CRITICAL  (MainThread) File "/Users/argonlab2/anaconda/lib/python2.7/site-packages/kiva/agg/agg.py", line 1012, in lines
    def lines(self, *args): return _agg.GraphicsContextArray_lines(self, *args)
root                                    : 2016-12-14 11:02:11,390 CRITICAL  (MainThread) TypeError: Cannot cast array data from dtype('O') to dtype('float64') according to the rule 'safe'
root                                    : 2016-12-14 11:02:11,390 CRITICAL  (MainThread) ============ End Exception ==========
ignore exception
 

```

wcmcintosh, 2016-12-14:
correct tab open, but no plot displayed

nmgrluser, 2016-12-14:
what analyses

closed 2017-04-19 by jirhiker