NMGRL/pychron #1057: error trying to access Toba Data

wcmcintosh, 2016-12-31:
```
SampleBrowserModel                      : 2016-12-31 10:58:23,246 DEBUG     (MainThread) filter by button fired low_post=None, high_post=None
Inspection                              : 2016-12-31 10:58:23,248 DEBUG     (MainThread) _load_associated_labnumbers called by _filter_by_hook. parent call=<pychron.envisage.browser.sample_browser_model.SampleBrowserModel object at 0x1284129b0> self._load_associated_labnumbers()
Inspection                              : 2016-12-31 10:58:23,249 DEBUG     (MainThread) _make_labnumbers called by _load_associated_labnumbers. parent call=<pychron.envisage.browser.sample_browser_model.SampleBrowserModel object at 0x1284129b0> sams.extend(self._make_labnumbers())
pychronmeta                             : 2016-12-31 10:58:23,249 DEBUG     (MainThread) ------- Get Labnumbers -------
pychronmeta                             : 2016-12-31 10:58:23,249 DEBUG     (MainThread) ------- principal_investigators: ['Chesner, C']
pychronmeta                             : 2016-12-31 10:58:23,249 DEBUG     (MainThread) ------- projects: ['Toba']
pychronmeta                             : 2016-12-31 10:58:23,249 DEBUG     (MainThread) ------- experiments: None
pychronmeta                             : 2016-12-31 10:58:23,249 DEBUG     (MainThread) ------- mass_spectrometers: []
pychronmeta                             : 2016-12-31 10:58:23,249 DEBUG     (MainThread) ------- irradiation: None
pychronmeta                             : 2016-12-31 10:58:23,249 DEBUG     (MainThread) ------- level: None
pychronmeta                             : 2016-12-31 10:58:23,249 DEBUG     (MainThread) ------- analysis_types: None
pychronmeta                             : 2016-12-31 10:58:23,250 DEBUG     (MainThread) ------- high_post: None
pychronmeta                             : 2016-12-31 10:58:23,250 DEBUG     (MainThread) ------- low_post: None
pychronmeta                             : 2016-12-31 10:58:23,250 DEBUG     (MainThread) ------------------------------
root                                    : 2016-12-31 10:58:23,250 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2016-12-31 10:58:23,250 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2016-12-31 10:58:23,251 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/traits/trait_notifiers.py", line 340, in __call__
    self.handler( *args )
root                                    : 2016-12-31 10:58:23,251 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_model.py", line 211, in _filter_by_button_fired
    self._filter_by_hook()
root                                    : 2016-12-31 10:58:23,251 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/base_browser_model.py", line 694, in _filter_by_hook
    self._load_associated_labnumbers()
root                                    : 2016-12-31 10:58:23,251 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/core/codetools/inspection.py", line 42, in dec
    return func(*args, **kw)
root                                    : 2016-12-31 10:58:23,251 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/base_browser_model.py", line 371, in _load_associated_labnumbers
    sams.extend(self._make_labnumbers())
root                                    : 2016-12-31 10:58:23,251 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/core/codetools/inspection.py", line 42, in dec
    return func(*args, **kw)
root                                    : 2016-12-31 10:58:23,251 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/base_browser_model.py", line 430, in _make_labnumbers
    ls = self._retrieve_labnumbers()
root                                    : 2016-12-31 10:58:23,251 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_model.py", line 315, in _retrieve_labnumbers
    filter_non_run=self.filter_non_run_samples)
root                                    : 2016-12-31 10:58:23,251 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc_database.py", line 1002, in get_labnumbers
    q = q.join(PrincipalInvestigatorTbl)
root                                    : 2016-12-31 10:58:23,251 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 1946, in join
    from_joinpoint=from_joinpoint)
root                                    : 2016-12-31 10:58:23,251 CRITICAL  (MainThread) File "<string>", line 2, in _join
root                                    : 2016-12-31 10:58:23,251 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/sqlalchemy/orm/base.py", line 201, in generate
    fn(self, *args[1:], **kw)
root                                    : 2016-12-31 10:58:23,252 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2089, in _join
    outerjoin, create_aliases, prop)
root                                    : 2016-12-31 10:58:23,252 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2157, in _join_left_to_right
    self._join_to_left(l_info, left, right, onclause, outerjoin)
root                                    : 2016-12-31 10:58:23,252 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2285, in _join_to_left
    "Tried joining to %s, but got: %s" % (right, ae))
root                                    : 2016-12-31 10:58:23,252 CRITICAL  (MainThread) InvalidRequestError: Could not find a FROM clause to join from.  Tried joining to <class 'pychron.dvc.dvc_orm.PrincipalInvestigatorTbl'>, but got: Can't find any foreign key relationships between 'IrradiationPositionTbl' and 'PrincipalInvestigatorTbl'.
root                                    : 2016-12-31 10:58:23,252 CRITICAL  (MainThread) ============ End Exception ==========
ignore exception

```

jirhiker, 2016-12-31:
already fixed in release/v16.12

closed 2016-12-31 by jirhiker

wcmcintosh, 2016-12-31:
Jake:

Thanks.

I think that I am operating release v16.7, apparently five releases back.

I’ll try to access 16.12.

I’m curious that my version keeps getting updates.

Sorry I’m not a little more on top of the versioning.

Bill

wcmcintosh, 2016-12-31:
Hi Jake:

	You have been closing issues like a fiend.

	Would it be possible to make the Pantelleria data available to Pychron?  Monitors and six unknowns from NM-264C and NM-278K.

Thanks,
Bill

jirhiker, 2016-12-31:
"I’m curious that my version keeps getting updates."

Its likely that the version that is being updated is not the version you are actually using. If you are launching from pycharm the "UpdatePlugin" is not relevant. (This is not strictly true but is true in your case. The UpdatePlugin updates source code hidden to the user in ~/.pychron/updates. When you launch from pycharm you are using source code located at /Users/mcintosh/Programming/git/pychron. If you open ~/.pychron/updates as a project in Pycharm then UpdatePlugin would be relevant.  This is what im planning on doing for Jan and Felix. This way I can view the dock version source code easily from pycharm. )