NMGRL/pychron #1063: tag status

wcmcintosh, 2017-04-13:
tag selecting still non-straight forward

even if data lines are actually selected, using set invalid sometimes changes selected lines, sometimes all "omit" lines, selected or no

tag icon brings up list of all omitted analyses, with selected analyses added on to end of list, in some cases duplicating omitted analyses at beginning of list.  If analyses are selected, should just these be listed if the tag icon is pushed?

tags are persisting much better now.  Tables seem to respect tags for making means, but analyses with "omit" tags do not match X in column one in spreadsheet.  Almost all data lines have X in column one, even omitted ones.   

Also, could tables show "n=" for mean values?

wcmcintosh, 2017-04-13:
in tables, the x's in column 1 are OK for the last data set, but not for the preceding sets

wcmcintosh, 2017-04-13:
When making tables, Pychron seems to default to run order for ordering the samples.  Particularly for large data sets, the user will want to order the samples in some particular way.  Is there any way to die this in the current pychron?  Ideally, one might want to build a list using append, but I don't see how to do this now.

jirhiker, 2017-04-18:
groups of analyses can be created in many ways

One easy on the fly method is to use the Browser Analysis Tables' "Freeze" feature. Select a set of analyses, right click> "Toggle Freeze" This will lock the selected analyses in the table so that when another sample is selected the selected analyses are not lost.

closed 2018-01-20 by jirhiker