NMGRL/pychron #1068: Error when selecting sample in browser

wcmcintosh, 2017-04-19:
Error when selecting sample in browser

All well after pychron restarted

I had had this error a few times

```
root                                    : 2017-04-19 11:47:03,566 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2017-04-19 11:47:03,566 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2017-04-19 11:47:03,566 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/traits/trait_notifiers.py", line 340, in __call__
    self.handler( *args )
root                                    : 2017-04-19 11:47:03,566 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_model.py", line 234, in _selected_samples_changed
    self._selected_samples_changed_hook(new)
root                                    : 2017-04-19 11:47:03,566 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/sample_browser_model.py", line 225, in _selected_samples_changed_hook
    self.analysis_table.set_analyses(ans, selected_identifiers={ai.identifier for ai in new})
root                                    : 2017-04-19 11:47:03,566 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/analysis_table.py", line 141, in set_analyses
    aa = [ai for ai in aa if ai.identifier in selected_identifiers]
root                                    : 2017-04-19 11:47:03,566 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/database/records/isotope_record.py", line 118, in __getattr__
    return getattr(self.dbrecord, item)
root                                    : 2017-04-19 11:47:03,566 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc_orm.py", line 161, in identifier
    return self.irradiation_position.identifier
root                                    : 2017-04-19 11:47:03,567 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/sqlalchemy/orm/attributes.py", line 237, in __get__
    return self.impl.get(instance_state(instance), dict_)
root                                    : 2017-04-19 11:47:03,567 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/sqlalchemy/orm/attributes.py", line 583, in get
    value = self.callable_(state, passive)
root                                    : 2017-04-19 11:47:03,567 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/sqlalchemy/orm/strategies.py", line 505, in _load_for_state
    (orm_util.state_str(state), self.key)
root                                    : 2017-04-19 11:47:03,567 CRITICAL  (MainThread) DetachedInstanceError: Parent instance <AnalysisTbl at 0x12ecbfb50> is not bound to a Session; lazy load operation of attribute 'irradiation_position' cannot proceed
root                                    : 2017-04-19 11:47:03,567 CRITICAL  (MainThread) ============ End Exception ==========
```

jirhiker, 2017-04-19:
I am aware of this type of error. It involves the database, sessions and ensuring data integrity. It's the root cause of most of the anxiety wrt data reduction and pychron "usability"

However since it does not occur consistently it would be nice to know the exact steps taken to produce the error.

wcmcintosh, 2017-04-20:
I will try to reproduce the error.

wcmcintosh, 2017-05-24:
I have not seen this problem in a while

closed 2017-10-19 by jirhiker