NMGRL/pychron #1080: question- review

wcmcintosh, 2017-04-25:
if I select one analysis then right click on review, i get a look at the review criteria, then that analysis has the green ball indicating review.  I can't review multiple analyses in one operation.

I would like to review all analyses.  I tried adding a review node - no error but execution halted

Is there any reason not to auto review all data in the process of data selection?

jirhiker, 2017-04-25:
"Is there any reason not to auto review all data in the process of data selection?"
calculating the review status is expensive so doing it blanketly for analyses would slow down pychron. Explicit is better than Implicit.

closed 2017-04-25 by jirhiker

wcmcintosh, 2017-04-25:
Is the review node functional?

jirhiker, 2017-04-25:
Yes. It functions the same way it always has. It is not a review of analyses, its is a simple pause in the pipeline giving the user the ability to examine what has occurred up to that point in the pipeline.