NMGRL/pychron #1081: spectrum pipeline - no output

wcmcintosh, 2017-04-25:
running spectrum pipeline, browser interaction OK, but no output

no error but loading failure error:

```
2017-04-25 14:50:39.786 python[10105:1774219] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
SampleBrowserModel                      : 2017-04-25 14:50:39,788 DEBUG     (MainThread) dump browser
SampleBrowserModel                      : 2017-04-25 14:50:39,798 DEBUG     (MainThread) $$$$$$$$$$$$$$$$$$$$$ Dumping browser selection
DVC                                     : 2017-04-25 14:50:39,807 DEBUG     (MainThread) sync repository Load1787. exists=True
Inspection                              : 2017-04-25 14:50:39,809 DEBUG     (MainThread) pull called by sync_repo. parent call=<pychron.dvc.dvc.DVC object at 0x127fb1890> repo.pull(use_progress=use_progress)
Load1787<GitRepo>                       : 2017-04-25 14:50:39,809 DEBUG     (MainThread) pulling master from origin
Load1787<GitRepo>                       : 2017-04-25 14:50:39,810 DEBUG     (MainThread) pulling from url: https://github.com/NMGRLData/Load1787.git
ISO                                     : 2017-04-25 14:50:40,549 WARNING   (MainThread)  - Failed loading peakcenter. error='data'
ISO                                     : 2017-04-25 14:50:40,555 WARNING   (MainThread)  - Failed loading peakcenter. error='data'
ISO                                     : 2017-04-25 14:50:40,560 WARNING   (MainThread)  - Failed loading peakcenter. error='data'
DVC                                     : 2017-04-25 14:50:40,574 DEBUG     (MainThread) Make analysis time, total: 0.767106056213, n: 14, average: 0.0547932897295
PipelineEngine                          : 2017-04-25 14:50:40,574 DEBUG     (MainThread) Pre run failed Unknowns<UnknownNode>
Pipeline Data Processing                : 2017-04-25 14:50:40,575 DEBUG     (MainThread) true run pipeline run_pipeline
Pipeline Data Processing                : 2017-04-25 14:50:40,575 DEBUG     (MainThread) run pipeline finished
```

wcmcintosh, 2017-04-25:
same problem for ideogram

jirhiker, 2017-04-25:
could not replicated problem

closed 2017-04-25 by jirhiker