NMGRL/pychron #1082: Spectrum - sigma display

wcmcintosh, 2017-04-25:
The sigma error display on the spectrum figure needs attention

ideograms correctly display how many sigma are used for mean and data

spectrum shows # sigma for age and error envelope.  However, the # sigma currently display for age is the number of sigma used for the plateau criteria, rather than number of sigma in plateau and integrated age
also. perhaps "Age±" should read "Plateau and integrated age ±"

jirhiker, 2017-04-25:
"spectrum shows # sigma for age and error envelope. However, the # sigma currently display for age is the number of sigma used for the plateau criteria, rather than number of sigma in plateau and integrated age" 
The code does not support this statement nor could I replicate the behavior

jirhiker, 2017-04-25:
the problem is that there are 3 nsigmas necessary and only 2 being specified

1. Plateau overlap n-sigma
2. Error Envelope display n-sigma
3. Reported Age n-sigma

currently 1 is also being used for 3

jirhiker, 2017-04-25:
OK. what i said is not true

Plateau overlap nsigma is always 2. 

so the user has control of Error Envelope and Reported Age nsigma

closed 2018-04-04 by jirhiker