NMGRL/pychron #1111: error saving intrerpreted age from IA ideogram

wcmcintosh, 2017-05-23:
active branch=None

active analyses=

```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/pyface/ui/qt4/action/action_item.py", line 343, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 333, in set_interpreted_age
    self._add_interpreted_ages(ias)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 473, in _add_interpreted_ages
    dvc.add_interpreted_age(ia)
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 743, in add_interpreted_age
    kca=float(ia.preferred_kca_value),
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 459, in _get_preferred_kca_value
    pa = self.preferred_kca
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 474, in _get_preferred_kca
    pa = self.weighted_kca
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 211, in _get_weighted_kca
    return ufloat(*self._calculate_weighted_mean('kca'))
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 269, in _calculate_weighted_mean
    return self._calculate_mean(attr, use_weights=True, error_kind=error_kind)
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 248, in _calculate_mean
    args = self._get_values(attr)
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 243, in _get_values
    vs, es = zip(*[(v.nominal_value, v.std_dev) for v in vs])
AttributeError: 'float' object has no attribute 'nominal_value'

```

jirhiker, 2017-05-23:
making IAs from IAs was never requested so its not necessarily going to work without modification

jirhiker, 2017-05-23:
see #1113

closed 2017-05-23 by jirhiker